ARMmbed/mbed-cli #861: ERROR: The target board you compiled for is not connected to your system

AnjanDhumal, 2019-03-01:
Hi,
I am having  a nordic nrf52 board connected 
> mbedls
| platform_name | platform_name_unique | mount_point | serial_port | target_id    | daplink_version |
|---------------|----------------------|-------------|-------------|--------------|-----------------|
| NRF52_DK      | NRF52_DK[0]          | E:          | COM6        | 000683585564 | unknown         |

However when i try to compile and flash it says 
> mbed compile --flash 
I get [mbed] ERROR: The target board you compiled for is not connected to your system.
       Please reconnect it and retry the last command.

AnjanDhumal, 2019-03-01:
description in comments

theotherjimmy, 2019-03-01:
@AnjanDhumal you could try reading the drive mentioned by `mbedls` (you can also use `mbed detect` to get the same info). `mbed flash` is probably trying to copy the build output onto that drive

-------

I also updated your comment to format the shell output as fixed width text. In the future you can use ` ``` ` on a line by itself before and after shell output to format it the same way (you can also click the edit button to view the changes I made). For what it's worth, `mbedls` output is also a markdown table when copied directly and prefixed with an empty line:

| platform_name | platform_name_unique | mount_point | serial_port | target_id    | daplink_version |
|---------------|----------------------|-------------|-------------|--------------|-----------------|
| NRF52_DK      | NRF52_DK[0]          | E:          | COM6        | 000683585564 | unknown         |

AnjanDhumal, 2019-03-04:
Yes I am able to detect using the commnad mbed detect 
mbed detect
[mbed] Working path ``` "D:\denised" (program)

[mbed] Detected NRF52_DK, port COM6, mounted E:, interface version unknown:
[mbed] Supported toolchains for NRF52_DK
| Target   | mbed OS 2 | mbed OS 5 | uARM |    IAR    |    ARM    |  GCC_ARM  |
|----------|-----------|-----------|------|-----------|-----------|-----------|
| NRF52_DK |     -     | Supported |  -   | Supported | Supported | Supported |
Supported targets: 1
Supported toolchains: 3 ```
Problem is with command >mbed compile flash 
Comiplation works fine but while copying the build output it fails with the error mentioned originally.

theotherjimmy, 2019-03-04:
@AnjanDhumal Could you try manually copying the Image file onto the disk? If that works, then We're going to need more information to track down this as a bug.

AnjanDhumal, 2019-03-04:
@theotherjimmy Yes, I am able to copy it manually onto the disk but not with this tool. Please let me know what information you would require.

theotherjimmy, 2019-03-04:
@AnjanDhumal if you're doing it with the tool that's not "manually". Hmmmmm... @bridadan How do we debug this?

theotherjimmy, 2019-03-04:
Oh, right that serial port being missing is the problem: Mbed CLI detects which board you have connected with serial port info.

@AnjanDhumal could you run `mbedls -v` and post the output?

AnjanDhumal, 2019-03-04:
mbedls -v
usage: mbedls [-h] [-s | -j | -J | -p | -P | -l | --version | -m ID:NAME]
              [--skip-retarget] [-u] [-d]
mbedls: error: unrecognized arguments: -v
Not sure if that was the output you were expecting

theotherjimmy, 2019-03-04:
You're correct, I was thinking of `-d`, so `mbedls -d`

Note to self, `-v` is common, so it should also work.

AnjanDhumal, 2019-03-04:
DEBUG:mbedls.main:mbed-ls ver. 1.7.2
DEBUG:mbedls.main:host: ('nt', 'Windows', '10', '10.0.14393', 'win32')
DEBUG:mbedls.lstools_win7:_iter_vals <PyHKEY object at 0x044CA1B0>
DEBUG:mbedls.lstools_win7:Mount point E: found for volume _??_USBSTOR#Disk&Ven_SEGGER&Prod_MSD_Volume&Rev_1.00#7&a09264f&0&000683585564&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}
DEBUG:mbedls.lstools_win7:Fetching mounted devices from disk service registry entry
DEBUG:mbedls.lstools_win7:_iter_vals <PyHKEY object at 0x04489C20>
DEBUG:mbedls.lstools_win7:Fetching usb storage devices from USBSTOR service registry entry
DEBUG:mbedls.lstools_win7:_iter_vals <PyHKEY object at 0x04489C20>
DEBUG:mbedls.lstools_win7:target_id_usb_id -> mount_point mapping: {u'000683585564': 'E:'}
DEBUG:mbedls.lstools_win7:Assigning new entry key string of 6&3732e29&0 to device 000683585564, as found in ParentIdPrefix
DEBUG:mbedls.lstools_win7:Composite device subdevice key SYSTEM\CurrentControlSet\Enum\USB\VID_1366&PID_1015&MI_01 was not found, skipping
DEBUG:mbedls.lstools_win7:Unknown capabilities from the following ids: [u'usb\\class_ff&subclass_ff&prot_ff', u'usb\\class_ff&subclass_ff', u'usb\\class_ff']
DEBUG:mbedls.lstools_win7:Composite device subdevice key SYSTEM\CurrentControlSet\Enum\USB\VID_1366&PID_1015&MI_04 was not found, skipping
DEBUG:mbedls.lstools_base:Candidates for display [{'mount_point': 'E:', 'product_id': u'1015', 'vendor_id': u'1366', 'serial_port': u'COM6', 'target_id_usb_id': u'000683585564'}]
DEBUG:mbedls.lstools_win7:Mount point E: is ready
DEBUG:mbedls.lstools_base:Could not read htm on from usb id 000683585564. Falling back to usb id
```
| platform_name | platform_name_unique | mount_point | serial_port | target_id    | daplink_version |
|---------------|----------------------|-------------|-------------|--------------|-----------------|
| NRF52_DK      | NRF52_DK[0]          | E:          | COM6        | 000683585564 | unknown         |
```
DEBUG:mbedls.main:Return code: 0

bridadan, 2019-03-04:
Hi @AnjanDhumal, thanks for providing all of this info. I have a few more questions.

- What is the output of `mbed config -G -L`?
- What version of Mbed CLI are you using? (`mbed --version`)
- What version of Mbed OS are you using? (`mbed ls` <- Note this is different from `mbedls`)

AnjanDhumal, 2019-03-05:
```
mbed config -G -L
[mbed] Working path "D:\denised" (program)
[mbed] Global config:
No global configuration is set

[mbed] Local config (D:\denised):
TOOLCHAIN=GCC_ARM
TARGET=NRF52840_DK
```
```
mbed --version
1.8.3
```
```
mbed ls
[mbed] Working path "D:\denised" (program)
denised (#63ac468d200e)
`- mbed-os (#ecb3c8c83716, tag: mbed-os-5.11.4)
```

bridadan, 2019-03-05:
Hi @AnjanDhumal, looks like you have your target set to `NRF52840_DK` instead of `NRF52_DK`.

Please run `mbed target NRF52_DK` and try flashing again.

AnjanDhumal, 2019-03-05:
Hi @bridadan  , that works but i thought NRF52840_DK was a supported MCU.

bridadan, 2019-03-05:
It definitely is, however it does not match what you have attached to your computer, which is why the tools alerted you that `The target board you compiled for is not connected to your system.`

bridadan, 2019-03-05:
Glad to hear it worked! I will close this now because the issue is solved, but let me know if you have further questions.

closed 2019-03-05 by bridadan

gracedotdotdot, 2020-03-06:
Hi there, I'm using board FRDM-K66F and is encountering the same question"The target board you compiled for is not connected to your system." but my mbed config -G -L is
[mbed] Global config:
GCC_ARM_PATH=/usr/bin/
MBED_OS_DIR=/home/ee2405/ee2405/mbed-os

[mbed] Local config (/home/ee2405/ee2405/mbed0/mbed-os-test-program):
TARGET=K66F