ARMmbed/mbed-cli #867: "mbed compile" 'str' object has no attribute 'decode'

jackyhuang85, 2019-03-07:
Same problem as #853. 
Using `Python 3.6.8` `mbed-cli 1.9.1`, after `mbed new` creating a new mbed project, 
`mbed compile -t GCC_ARM -m K66F` appears following error message:
```
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "/Users/jackyhuang/GoogleDrive/TA/Embedded_system/test" (program)
[Warning] @,: Compiler version mismatch: Have 7.2.1; expected version >= 6.0.0 and < 7.0.0
Building project test (K66F, GCC_ARM)
Scan: test
Compile [  3.6%]: MbedCRC.cpp
[ERROR] 'str' object has no attribute 'decode'
[mbed] ERROR: "/Users/jackyhuang/miniconda3/bin/python" returned error.
       Code: 1
       Path: "/Users/jackyhuang/GoogleDrive/TA/Embedded_system/test"
       Command: "/Users/jackyhuang/miniconda3/bin/python -u /Users/jackyhuang/GoogleDrive/TA/Embedded_system/test/mbed-os/tools/make.py -t GCC_ARM -m K66F --source . --build ./BUILD/K66F/GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
---
[mbed] WARNING: Using Python 3 with Mbed OS 5.8 and earlier can cause errors with compiling, testing and exporting
---
```

As mentioned in https://github.com/ARMmbed/mbed-os/pull/9673  , 
this bug "may" be fixed in Mbed os 5.11.5? But today it happens again.

Interestingly, when `mbed import mbed-os-example-blinky` and compile process can be completed.
Both are using Mbed os 5.11.5

jackyhuang85, 2019-03-07:
I found the problem!
when `mbed new` , it pull the "latest" tag version from [ARMmbed/mbed-os](https://github.com/ARMmbed/mbed-os) but the "latest" tag not changed to 5.11.5, still 5.11.4
```
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at branch/tag "latest"
```

Whereas, `mbed import mbed-os-example-blinky` pulls #51d55508e840 commit, which is the release version 5.11.5
```
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #51d55508e840
```
Report ARMmbed/mbed-os ?

bridadan, 2019-03-07:
Thanks for raising this @jackyhuang85, looks like the issue has been solved on the mbed-os repo.

closed 2019-03-07 by bridadan