ARMmbed/mbed-cli #869: Running in Python 3 attempts to use raw_input

arskanov, 2019-03-14:
Hi,

I ran into this issue while trying out mbed-cli 1.9.1 on python3.5. Mbed-OS version 5.11.5.
Script fails in the following way:
```
> mbed publish
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "/home/aramanen/work/riotIisy" (program)
[mbed] Checking for local modifications...
[mbed] Uncommitted changes in program "riotIisy" in "/home/aramanen/work/riotIisy"
[mbed] ERROR: Unknown Error: name 'raw_input' is not defined
---
[mbed] WARNING: Using Python 3 with Mbed OS 5.8 and earlier can cause errors with compiling, testing and exporting
---
```

The error is on this [this line](https://github.com/ARMmbed/mbed-cli/blob/3b43228d6f21d99a4919866c697c0f73cecee313/mbed/mbed.py#L2329), and fails since raw_input() is input() in py3.
```
    if repo.dirty():
        action("Uncommitted changes in %s \"%s\" in \"%s\"" % (repo.pathtype(repo.path), repo.name, repo.path))
        if msg:
            repo.commit(msg)
        else:
            if sys.version_info[0] == 3:
                input('Press enter to commit and publish: ')
            else:
                raw_input('Press enter to commit and publish: ')
            repo.commit()
```
Seems to make it work fine.

arskanov, 2019-03-14:
Steps to reproduce:
1. Install module via pipenv
2. clone blinky
3. Modify two files in the repository
4. Commit one of them via git, leave the other as unstaged
5. run `mbed publish`

Should mbed-cli be able to handle this situation?

Sorry about the horrible shell formatting.
```

 ~/tmp  mkdir mcl2
 ~/tmp  cd mcl2
 ~/t/mcl2  pipenv --three
Creating a virtualenv for this project…
Pipfile: /home/aramanen/tmp/mcl2/Pipfile
Using /usr/bin/python3.5m (3.5.2) to create virtualenv…
⠋ Creating virtual environment...Using base prefix '/usr'
New python executable in /home/aramanen/.local/share/virtualenvs/mcl2-BkELMTBZ/bin/python3.5m
Also creating executable in /home/aramanen/.local/share/virtualenvs/mcl2-BkELMTBZ/bin/python
Installing setuptools, pip, wheel...
done.
Running virtualenv with interpreter /usr/bin/python3.5m

✔ Successfully created virtual environment! 
Virtualenv location: /home/aramanen/.local/share/virtualenvs/mcl2-BkELMTBZ
Creating a Pipfile for this project…
 ~/t/mcl2  pipenv install mbed-cli==1.9.2
Installing mbed-cli==1.9.2…
Adding mbed-cli to Pipfile's [packages]…
✔ Installation Succeeded 
Pipfile.lock not found, creating…
Locking [dev-packages] dependencies…
Locking [packages] dependencies…
✔ Success! 
Updated Pipfile.lock (1ca9e4)!
Installing dependencies from Pipfile.lock (1ca9e4)…
  🐍   ▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉ 1/1 — 00:00:00
To activate this project's virtualenv, run pipenv shell.
Alternatively, run a command inside the virtualenv with pipenv run.
 ~/t/mcl2  pipenv shell
Launching subshell in virtual environment…
 ~/t/mcl2   source /home/aramanen/.local/share/virtualenvs/mcl2-BkELMTBZ/bin/activate.fish
(mcl2)  mcl2-BkELMTBZ  ~/t/mcl2  mbed-cli --version
1.9.2
(mcl2)  mcl2-BkELMTBZ  ~/t/mcl2  mbed import https://github.com/arskanov/mbed-os-example-blinky.git
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "/home/aramanen/tmp/mcl2" (directory)
[mbed] Importing program "mbed-os-example-blinky" from "ssh://git@github.com/arskanov/mbed-os-example-blinky" at latest revision in the current branch
[mbed] Adding library "mbed-os" from "ssh://git@github.com/ARMmbed/mbed-os" at rev #51d55508e840
[mbed] Auto-installing missing Python modules (colorama, urllib3, prettytable, junit_xml, pyyaml, jsonschema, future, six, mbed_cloud_sdk, requests, idna, pyserial, jinja2, intelhex, intervaltree, mbed_ls, mbed_host_tests, mbed_greentea, beautifulsoup4, fuzzywuzzy, pyelftools, manifest_tool, icetea)...
(mcl2)  mcl2-BkELMTBZ  ~/t/mcl2  
(mcl2)  mcl2-BkELMTBZ  ~/t/mcl2  cd mbed-os-example-blinky/
(mcl2)  mcl2-BkELMTBZ  ~/t/m/mbed-os-example-blinky   master   echo "#BIG" >> README.md
(mcl2)  mcl2-BkELMTBZ  ~/t/m/mbed-os-example-blinky   master ±  echo "// Comment" >> main.cpp
(mcl2)  mcl2-BkELMTBZ  ~/t/m/mbed-os-example-blinky   master ±  git add main.cpp
(mcl2)  mcl2-BkELMTBZ  ~/t/m/mbed-os-example-blinky   master ±  git commit -m "Add comment"
[master e819067] Add comment
 1 file changed, 1 insertion(+)
(mcl2)  mcl2-BkELMTBZ  ~/t/m/mbed-os-example-blinky   master ±  gs
On branch master
Your branch is ahead of 'origin/master' by 1 commit.
  (use "git push" to publish your local commits)
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   README.md

no changes added to commit (use "git add" and/or "git commit -a")
(mcl2)  mcl2-BkELMTBZ  ~/t/m/mbed-os-example-blinky   master ±  mbed status
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "/home/aramanen/tmp/mcl2/mbed-os-example-blinky" (program)
[mbed] Status for "mbed-os-example-blinky":
 M README.md

(mcl2)  mcl2-BkELMTBZ  ~/t/m/mbed-os-example-blinky   master ±  mbed publish
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "/home/aramanen/tmp/mcl2/mbed-os-example-blinky" (program)
[mbed] Checking for local modifications...
[mbed] Uncommitted changes in program "mbed-os-example-blinky" in "/home/aramanen/tmp/mcl2/mbed-os-example-blinky"
[mbed] ERROR: Unknown Error: name 'raw_input' is not defined
---
[mbed] WARNING: Using Python 3 with Mbed OS 5.8 and earlier can cause errors with compiling, testing and exporting
---
(mcl2)  ✘  mcl2-BkELMTBZ  ~/t/m/mbed-os-example-blinky   master ±  
```

theotherjimmy, 2019-03-14:
Resolved by #870 Good job!

closed 2019-03-14 by theotherjimmy