ARMmbed/mbed-cli #871: Files/folders built but not used; mbedignore

mjohngreene, 2019-03-17:
Each build run using the CLI takes quite a while obviously depending partly on the power of the PC/machine used.  Another primary contributor to the delay, though, is that many extraneous files seem to be compiled during the build but never even used for linking/assembly to produce the resultant executable.  We are creating and otherwise trying to more effectively use mbedignore.  Are there any tools/methods to readily determine which compiled files are not ultimately used in the executable?  Any guidance/linkage to instructions/explanations would be greatly appreciated.

theotherjimmy, 2019-03-18:
@mjohngreene They are used; Those files are all passed to the linker and the linker sorts them out. The memap output can probably guide you to figure out what is actually used and what is garbage collected by the linker. Be aware that using `.mbedignore` files to remove code for quicker `-c` builds may lead to odd symbol not defined errors at the link step that can be hard to track down.

Please be sure that you're not running compiles with `-c`. If that does not re-build when it's required, please report that separately.

mjohngreene, 2019-03-20:
Thanks, @theotherjimmy.  Is there a good example `.mbedignore` (ie, one with more than a few lines of content) available?

theotherjimmy, 2019-03-20:
@mjohngreene `.mbedignore` is probably not a great solution to this problem, and becomes a maintenance nightmare after a few lines. I don't have an example.

mjohngreene, 2019-03-20:
OK, thanks

closed 2019-03-20 by mjohngreene