ARMmbed/mbed-cli #872: How to config arm library in mbed

ccu1tn, 2019-03-18:
Dear everyone;
I have a c++ file like this 
[image]
 When i use command line "mbed compile" to compiler that file. It has errors 
[image]
Am I configuring wrong the arm library in mbed?
Please, everyone give me some idea to fix it.
Many thank

theotherjimmy, 2019-03-18:
@ccu1tn I think you may be confusing Mbed OS and CMSIS stuff. `RTE_Components.h` is generated by uVision for CMSIS using projects, `EventRecorder.h` is part of the Arm compiler 6 pack, etc.

ccu1tn, 2019-03-18:
Dear @theotherjimmy ;
So, How to I can fix those errors? How can I compile that program is not errors?

theotherjimmy, 2019-03-18:
@ccu1tn I'm not here to do your homework for you so I can't answer your exact question. Please use the documentation as os.mbed.com/docs and the APIs there. You're using APIs described in https://arm-software.github.io/CMSIS_5/General/html/index.html. Mbed CLI compiles Mbed OS projects using the Mbed OS APIs

ccu1tn, 2019-03-19:
Dear @theotherjimmy 
It is not my homework. I make compile it. but it has that errors... I already read and take follow this document but it still errors. 
Because you have an experiment for it. can you give me the command line to compile is correct?

Many thank

40Grit, 2019-03-19:
Hoping not to disrespect. But the more I see near nonsensical issues and MRs like this the more I worry that research AIs have been unleashed on GitHub.

theotherjimmy, 2019-03-19:
@ccu1tn The command line to compile will not change. You have to change the structure of your project so that it will compile. In particular, you have to be sure that your project contains the files your including. In particular, using `find` in conjunction with the [Build Rules Document](https://os.mbed.com/docs/mbed-os/v5.11/reference/mbed-os-build-rules.html) may yield results fast

theotherjimmy, 2019-03-19:
@40Grit This issue is not nonsensical, so either the AIs are doing a great job or someone really is having issues compiling code that uses CMSIS from an Mbed OS project.

40Grit, 2019-03-20:
Yeah looking deeper, I guess I should have chose a different issue to unveil my conspiracy theory in.

closed 2019-04-18 by ciarmcom