ARMmbed/mbed-cli #875: mbed complie can't find ARM complier 6 in the PATH with quotes

jamesbeyond, 2019-03-22:
### Description

While using mbed-cli with ARMC6. I found a strange behaviour.
based on [this doc](https://os.mbed.com/docs/mbed-os/v5.11/tools/after-installation-configuring-mbed-cli.html), I can just put ARMC6 in the system PATH env var, then use it from there.

`mbed compile -t ARM -m K64F` 

If I do 
`set PATH=%PATH%;c:\Program Files\ARMCompiler6.11\bin`
mbed compile runs just fine. however if I run with quotes around ARMC6 bins
`set PATH=%PATH%;"c:\Program Files\ARMCompiler6.11\bin"`
mbed compile fail to detect ARMC6

and throw errors:
```
 [mbed] Working path "c:\mbed\examples\mbed-os-psa-compliance-tests-example" (program)
usage: make.py [-h] [-m MCU] [--custom-targets CUSTOM_TARGETS_DIRECTORY]
               [-t TOOLCHAIN] [--color] [--cflags CFLAGS]
               [--asmflags ASMFLAGS] [--ldflags LDFLAGS] [-c]
               [--profile PROFILE] [--app-config APP_CONFIG]
               [-p PROGRAM | -n PROGRAM | -L | -S [{matrix,toolchains,targets}]]
               [-j JOBS] [-v] [--silent] [-D MACROS] [-f GENERAL_FILTER_REGEX]
               [--stats-depth STATS_DEPTH] [--automated] [--host HOST_TEST]
               [--extra EXTRA] [--peripherals PERIPHERALS]
               [--dep DEPENDENCIES] [--source SOURCE_DIR]
               [--duration DURATION] [--build BUILD_DIR] [-N ARTIFACT_NAME]
               [--ignore IGNORE] [-b BAUD] [--rpc] [--usb] [--dsp] [--testlib]
               [--build-data BUILD_DATA] [-l LINKER_SCRIPT]
make.py: error: Could not find executable for ARMC6.
Currently set search path: No path set
[mbed] ERROR: "c:\python27\python.exe" returned error.
       Code: 2
       Path: "c:\mbed\examples\mbed-os-psa-compliance-tests-example"
       Command: "c:\python27\python.exe -u c:\mbed\examples\mbed-os-psa-compliance-tests-example\mbed-os\tools\make.py -D MBEDTLS_USER_CONFIG_FILE="../comp_config.h" -t ARM -m LPC55S69_S --profile release --source . --build .\BUILD\LPC55S69_S\ARM-RELEASE --app-config mbed-os/tools/psa/tfm/mbed_app.json"
       Tip: You could retry the last command with "-v" flag for verbose output
```
mbed-cli version 1.9.2

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

theotherjimmy, 2019-03-22:
@jamesbeyond This looks like it might be a bug with windows environment variables. What does `echo %PATH%` after the path with quotes give you?

jamesbeyond, 2019-03-25:
Hi @theotherjimmy,
The broken one shows
```
C:\Program Files\Microsoft VS Code\bin;C:\Users\qinshi01\AppData\Roaming\npm;C:\Program Files (x86)\Nmap;"c:\Program Files\ARMCompiler6.11\bin"
```
Which also include the quotes in the PATH Variable

theotherjimmy, 2019-03-25:
@jamesbeyond Then don't use quotes in the path variable.

jamesbeyond, 2019-03-25:
Hi @theotherjimmy,
I think this is not about how I use the tools. this is about customer experiences.
This gives customers the impression that our tools are fundamentally broken. Because they are following the instructions, and use complete valid settings but end up with unusable tools.
Arm compiler 6 works complete fine with quotes in the `PATH`, 
```
C:\>set PATH=%PATH%;"c:\Program Files\ARMCompiler6.11\bin"
C:\>armclang --help
Product: ARM Compiler 6.11 Professional
Component: ARM Compiler 6.11
Tool: armclang [5d3b4200]
```
our customer will be struggle to find-out what went wrong with mbed-cli, as it is complains it can't find `armclang`

bridadan, 2019-03-25:
It's quite common to wrap paths in Windows with quotes, so I'm with @jamesbeyond on this one. I don't think it should be to difficult to strip the quotes.

@jamesbeyond I can't work on a solution right this second due to activities around the upcoming release. However if you have time (and are motivated 😄) I'd be happy to point you in the direction for a patch. I think the appropriate place to do the proper clean up of toolchain paths would happen here: https://github.com/ARMmbed/mbed-os/blob/e69aa15aea92a6d7803280fde2dd9682fb8b0d92/tools/settings.py#L102. You would want to loop over the paths similar to what is done above and do the proper normalization.

bridadan, 2019-04-05:
Fix up here: https://github.com/ARMmbed/mbed-os/pull/10333

closed 2019-04-09 by 0xc0170