ARMmbed/mbed-cli #877: Compilation detect wrong target

lilrom13, 2019-03-24:
When I try to flash NRF52840-DK board I get the following error :

> ERROR: The target board you compiled for is not connected to your system.
>        Please reconnect it and retry the last command.

$ mbed compile -m detect

> [mbed] Detected "NRF52_DK" connected to "/Volumes/JLINK" and using com port "/dev/tty.usbmodem0006839251911"

bridadan, 2019-03-25:
Hi @lilrom13,

I don't have these boards on my desk so I can't reproduce this problem at the moment. However I remember looking at this sometime ago and I believe the JLink firmware is the same for both the NRF52840 and the NRF52_DK. This means we can't use the USBID or files present on the device's disk to distinguish these two platforms from each other.

There are two workarounds I can think of:

1. Mock that specific target id to show up as `NRF52840_DK`: https://github.com/ARMmbed/mbed-os-tools/tree/master/packages/mbed-ls#mocking-renaming-platforms
1. Replace the JLink firmware with DAPLink: https://armmbed.github.io/DAPLink/
    1. DAPLink should properly distinguish the two types of targets from each other

Let me know if you have any further questions!

cmonr, 2019-04-09:
Closing since this appears to be resolved.

closed 2019-04-09 by cmonr