ARMmbed/mbed-cli #878: mbed cli 1.10.0 - wrong .bss values using ARMC compiler

JanneKiiskila, 2019-03-25:
```
@ubuntu:~/mbed/mbed-client-lite-example-internal$ mbed --version
1.10.0
```
ARM C compiler 6.11 in use, running Ubuntu 16.04 LTS.

The reported RAM memory consumption is completely out of whack, it's way too large to be real.

```
Space used after regions merged: 0x34094
| Module                              |        .text |   .data |        .bss |
|-------------------------------------|--------------|---------|-------------|
| [lib]/c_w.l                         |  19013(+274) |  16(+0) |     348(+0) |
| [lib]/fz_wm.l                       |     1762(+0) |   0(+0) |       0(+0) |
| [lib]/libcpp_w.l                    |        1(+0) |   0(+0) |       0(+0) |
| [lib]/libcppabi_w.l                 |       44(+0) |   0(+0) |       0(+0) |
| [lib]/m_wm.l                        |      760(+0) |   0(+0) |       0(+0) |
| anon$$obj.o                         |       32(+0) |   0(+0) | 171584(+80) |
| drivers/wifi-x-nucleo-idw01m1       |      128(+0) |   0(+0) |       0(+0) |
| main.o                              |   1422(+602) |   0(+0) |      28(+4) |
| mbed-cloud-client/mbed-client       |  42387(+254) |  21(+0) |     485(+0) |
| mbed-cloud-client/source            |     2580(+0) |   5(+0) |      24(+0) |
| mbed-cloud-client/update-client-hub |    20936(+0) | 220(+0) |    5214(+0) |
| mbed-os/cmsis                       |     1021(+0) |   0(+0) |       0(+0) |
| mbed-os/components                  |    13329(+0) |   0(+0) |    3072(+0) |
| mbed-os/drivers                     |     4474(+0) |   0(+0) |      48(+0) |
| mbed-os/events                      |     2303(+0) |   0(+0) |    4424(+0) |
| mbed-os/features                    |    58531(+0) | 161(+0) |    1080(+0) |
| mbed-os/hal                         |     1808(+0) |   4(+0) |      66(+0) |
| mbed-os/platform                    |   6728(-420) |  80(+0) |   2773(-76) |
| mbed-os/rtos                        |    13313(+0) | 168(+0) |    6634(+0) |
| mbed-os/targets                     |    18870(+0) |  12(+0) |    1157(+0) |
| mbed_cloud_dev_credentials.o        |      265(+0) |   0(+0) |       0(+0) |
| source/application_init.o           |       96(+0) |   0(+0) |       0(+0) |
| source/blinky.o                     |    836(+108) |   1(+0) |       4(+0) |
| source/oma_lwm2m_object_defs.o      |      338(+0) |   0(+0) |       0(+0) |
| source/platform                     |  1098(-1318) |   0(+0) |     114(+0) |
| source/resource.o                   |    224(+224) |   0(+0) |       0(+0) |
| update_default_resources.o          |       41(+0) |   0(+0) |       0(+0) |
| Subtotals                           | 212340(-276) | 688(+0) |  197055(+8) |
Total Static RAM memory (data + bss): 197743(+8) bytes
Total Flash memory (text + data): 213028(-276) bytes

Update Image: ./BUILD/K64F/ARM/mbed-client-lite-example-internal_update.bin
Image: ./BUILD/K64F/ARM/mbed-client-lite-example-internal.bin

```

It should be really in the ballpark of 19-20 kB maximum. Seems this strange anon object is give a lot of RAM in this report for some reason.

bridadan, 2019-03-25:
I believe at one point @deepikabhavnani mentioned this may be the heap being picked up, but I don't know for sure. But agreed it should not be showing that there.

deepikabhavnani, 2019-04-09:
@bridadan - Yes heap region is picked up as part of bss

bulislaw, 2019-05-07:
Closing it as the reported memory is used for heap.

closed 2019-05-07 by bulislaw

JanneKiiskila, 2019-05-07:
It's not being reported the same way for other targets, so I think the closure is a bit premature. All compilers and targets should behave the same, I should not be getting this much different numbers from one (IAR) compared to the others.

bridadan, 2019-05-07:
I agree with @JanneKiiskila, the memory usage reported in ARMC6 should be consistent with GCC_ARM and IAR.