ARMmbed/mbed-cli #879: Print path of source file during compile

paradajz, 2019-04-01:
Currently, mbed compile will output only the filename of the file being compiled, ie.

```
...
Compile [ 20.1%]: cipher_wrap.c
...
```

But later, when the build is finished, full path relative to the source directory is printed:

`| mbed-os/features/mbedtls/src/cipher_wrap.o                                   |    826 |     0 |    0 |`

If there an option to print the full path during compilation as well? If not, consider this a feature request. My usecase is the following:

I am using mbed-compile in Visual Studio Code. I wrote regex matcher which will catch any warnings or errors during compilation, so that I get screen like this in Visual Studio Code's problems tab:

[image]

The problem is when I click on reported problem, Visual Studio Code doesn't know the exact path of the file, since the assumption is $CURRENT_PROJ_PATH + filename caught during compile. Since compile only gives out filename without path, this won't work properly (the built path will be invalid).

theotherjimmy, 2019-04-01:
Hi @paradajz. That's a good catch, and this could certainly be improved for consistency.  Perhaps you could be parsing the warning information directly instead of relying on progress info to print out exactly below their respective files. For reference, this is not always the case. progress and warnings may be intermingled at the moment.

paradajz, 2019-04-01:
Yes, I've been considering just turning the verbose output on during the compile and catching the warnings/errors from GCC although this feels like a hack since the mbed-cli already uses different syntax. Having the file path in mbed-cli syntax would be much more elegant solution.

theotherjimmy, 2019-04-02:
@paradajz As I said, that's not a good way to do things as I can't make certain that the errors/warnings always print after the file that generates them. We compile many files in parallel and we may print the "Compiling" for a file long before the warnings and many other "Compiling" statuses may appear between the two, making this an unreliable method of determining the source of an error or warning.

closed 2019-10-31 by ciarmcom