ARMmbed/mbed-cli #880: mbed sterm needs mbed-os project

coisme, 2019-04-02:
I installed Mbed CLI into new Windows 10 environment with executable binary `Mbed_installer_v0.4.10.exe`. Right after that, I tried connecting to K64F via serial terminal using `mbed sterm` command. The command resulted in error. 

However, if I import mbed-os, the command works. It's not intuitive to me that importing mbed-os is necessary just to use serial terminal.

```
C:\Users\Owner>mbed sterm -b 115200
[mbed] Working path "C:\Users\Owner" (directory)
[mbed] Program path "C:\Users\Owner"
[mbed] Detecting connected targets/boards to your system...
[mbed] ERROR: Couldn't detect connected targets/boards to your system.
       You can manually specify COM port via the '--port' option.
---

C:\Users\Owner>mbed import https://github.com/ARMmbed/mbed-os.git
[mbed] Working path "C:\Users\Owner" (directory)
[mbed] Program path "C:\Users\Owner"
[mbed] Importing program "mbed-os" from "https://github.com/ARMmbed/mbed-os.git" at latest revision in the current branch
[mbed] Auto-installing missing Python modules (colorama, urllib3, prettytable, junit_xml, pyyaml, jsonschema, future, six, mbed_cloud_sdk, requests, idna, pyserial, jinja2, intelhex, mbed_ls, mbed_host_tests, mbed_greentea, beautifulsoup4, pyelftools, manifest_tool, icetea, pycryptodome, pyusb, cmsis_pack_manager)...

C:\Users\Owner>mbed sterm -b 115200
[mbed] Working path "C:\Users\Owner" (directory)
[mbed] Program path "C:\Users\Owner"
[mbed] Detecting connected targets/boards to your system...
[mbed] Opening serial terminal to "K64F"
--- Terminal on COM4 - 115200,8,N,1 ---
ato connect to network (-3014). Retrying...                                <-- Serial output
Unable to connect to network (-3014). Retrying...                     <-- Serial output
Unable to connect to network (-3014). Retrying...                     <-- Serial output
Unable to connect to network (-3014). Retrying...                     <-- Serial output
```

theotherjimmy, 2019-04-02:
@coisme This is true of many other Mbed CLI subcommands. `mbed compile` does not work outside an mbed-os project either.

coisme, 2019-04-03:
@theotherjimmy I understand that `mbed compile` needs mbed-os because it compiles an Mbed OS project. However, `mbed sterm` is a serial terminal. In addition, my experiment above indicates `mbed-os` repository itself is not required. I'd like to request that `mbed sterm` works with only installing Mbed CLI because it is a nice feature for beginners.

For example, let's think about a beginner developing a Mbed OS program on online compiler. Usually, he need to see the serial output from his target board to check if the program works correctly. At that time, if `mbed sterm` works just after installing Mbed CLI from executable, it is very easy for that user. Otherwise, user have to care about OS, terminal software, and a COM port.

linlingao, 2019-04-03:
We'll treat this as a feature request then.

coisme, 2019-04-05:
@linlingao Ok, thank you!

bridadan, 2019-04-09:
Hey @coisme, I took at look at this. Turns out `mbed sterm` doesn't require `mbed-os` to be cloned locally, but it does need `mbed-ls` to be installed if you do not supply the `--port` argument. Just run `pip install mbed-ls` to install it, then it should work.

coisme, 2019-04-10:
Hi @bridadan, thank you for your research. Is it possible to include mbed-ls in the installer package? It is ideal for beginners that installing .exe is the only requirement for using `mbed sterm`.

bridadan, 2019-04-11:
Seems reasonable to me.

@theotherjimmy how do you feel about adding `mbed-ls` and `pyserial` as python dependencies for Mbed CLI? Since it directly imports them?

theotherjimmy, 2019-04-11:
I think it's the correct thing to do @bridadan.

PR #884 (2019-04-11): Add requirements
Fixes #880.

Both `pyserial` and `mbed-ls` (now using the `mbed-os-tools` package) are called directly by Mbed CLI, so let's add them as Python requirements.

Because `mbed-os-tools` is now a direct requirement, we can remove the `try/except` block to catch the import error.
closed 2019-04-11 by bridadan