ARMmbed/mbed-cli #881: mbed dm init always chooses the first developer certificate

ryannowarm, 2019-04-05:
It would be nice if mbed dm init allows the user to select which developer certificate gets downloaded.

bridadan, 2019-04-05:
Thanks for the feedback! At the moment, we generally don't prompt users after running a command. Would selecting a certificate by name with a `--cert-name` option or something be workable?

ryannowarm, 2019-04-05:
On Fri, Apr 05, 2019 at 08:59:30AM -0700, Brian Daniels wrote:
> Thanks for the feedback! At the moment, we generally don't prompt users after running a command. Would selecting a certificate by name with a `--cert-name` option or something be workable?
>

Hey Brian.  Yup a command line option would be best.  Probably something like
`--dev-cert-name` for clarity since there's already a `--certificate
CERTIFICATE` option to specify the manifest signing cert.
IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

closed 2019-10-31 by ciarmcom