ARMmbed/mbed-cli #882: Export IDE project (SW4STM32 and others) with either full sources, reduced source (as with online compiler) or library

Hoel, 2019-04-07:
We use the CLI export function to generate project for IDE (SW4STM32 in this case). There are however major issues with the resulting project :
-it is including all possible targets from all vendors, even tho only one target is actually used 
-it is then building with all possible features and option enabled / included, and rebuild everything at each compilation (which takes tens and tens of seconds). Also all these features/options are very hard to disable later on (have to add .mbedignore files everywhere and mess with project includes, it can literally take hours, and even tho for a same project the resulting binary is over twice larger than when compiled with online compiler..)
-there is no way with CLI to export a project which only contain the compiled mbed library rather than all sources

All these missing function have been implemented already in the online compiler, so it is kinda sad that the CLI miss them, especially because the current project resulting from export function is almost unusable, and on contrary the project exported from online compiler is working well.

In our case we work in SW4STM32 on custom targets (STM32L151CC and STM32L451CE/L462CE), so we cannot use the online compiler to generate a SW4STM32 project template (online compiler doesnt allow custom targets), we have no choice but to use the CLI export function. 

So, please, add the following options in CLI export tool:
-Export with minimal sources, meaning mbed sources with all non explicitely used features and option disabled, and only current target files present in "target" folder. (: exactly as the online compiler does) 
-export with compiled library only, meaning mbed headers + compiled library only.

theotherjimmy, 2019-04-08:
@Hoel Use `-z` to get a zip file like the online compiler does.

theotherjimmy, 2019-04-08:
@Hoel You cannot get a compiled library as Mbed OS is not intended to be built as a single `.a` or `.ar` file. That's because:
 1) There are weak symbols
 2) The configuration system uses macros, which would require a recompile.

Hoel, 2019-04-08:
@theotherjimmy
OK thank you, indeed i use -z and that's good enough, its what the online compiler provide when we export a project.

> You cannot get a compiled library

OK i see

Closing this issue then. 
There is still an hardfault issue with the generated SW4STM32 project, the same project built from CLI works perfectly  https://github.com/ARMmbed/mbed-os/issues/10339

closed 2019-04-08 by Hoel