ARMmbed/mbed-cli #883: Mbed CLI 0.4.10 will not install on a Win 10 system

woobagooba, 2019-04-10:
Installer appears to reach successful finish.  Mbed CLI not operable at command line.

I do have a second Win 10 system upon which the installer works fine.

How do you recommend I debug this issue on the problem system?

j3hill, 2019-04-11:
@woobagooba Did you run the Window Mbed-CLI installer located here: https://github.com/ARMmbed/mbed-cli-windows-installer/releases/tag/v0.4.10? Also are you running 64-bit Windows 10 (with recent updates, perhaps)? To fix the issue you can manually run: "pip install mbed-cli" from the command line. With that said, we need to determine why the install did not work as expected.

woobagooba, 2019-04-11:
> @woobagooba Did you run the Window Mbed-CLI installer located here: https://github.com/ARMmbed/mbed-cli-windows-installer/releases/tag/v0.4.10? 

Yes

> Also are you running 64-bit Windows 10 (with recent updates, perhaps)?

Yes.

linlingao, 2019-04-11:
@woobagooba When you create an issue, you should fill in the template with information you can provide, but leave the template intact.

woobagooba, 2019-04-11:
We have just spent the better part of a week fighting through several frustrations with both mbed-cli and mbed OS simulator.  You are complaining about the format in which we are providing feedback?

linlingao, 2019-04-11:
@woobagooba Leaving the format intact will help us understand the severity and set priority. Given your frustrations, surely you want that?

woobagooba, 2019-04-11:
Template, format ... what are you talking about?  Point to the forum guidelines / instructions please.

linlingao, 2019-04-11:
@woobagooba Sorry, my bad, this repo doesn't have template installed for some reason.

woobagooba, 2019-04-11:
No problem.  Now, instead of nit picking issue reporting formats, it would be much appreciated if instead  ARM could direct your considerable technical expertise to resolve this issue (and the related MBed OS Simulator issue).   We thank you.

j3hill, 2019-04-12:
@woobagooba We are still looking at this issue, but it currently appears to be related to the Python installation. If you run the mbed-cli uninstaller, then uninstall Python, and run the Mbed-CLI installer once again, it seems to work correctly. Are you running 64-bit Python? Did you install Python before installing the Mbed-CLI?

woobagooba, 2019-04-12:
@j3hill Thanks so much for your attention.  The Mbed-cli install issue arose was we were trying to install the Mbed OS simulator.  Those installation instructions were problematic, so in the spirit of trying to help the community we are authoring what we believe are a complete and precise set of MBed OS sim install  instructions.

> It is assumed git is installed on your execution PATH
> (Do not install node JS)
> Install Mbed CLI using instructions here ... 
> https://os.mbed.com/docs/mbed-os/v5.12/tools/installation-and-setup.html
> Install the Emscripten SDK using instructions from here ... 
> https://emscripten.org/docs/getting_started/downloads.html
> Choose a root folder for the mbed-simulator.
> git clone https://github.com/janjongboom/mbed-simulator.git
> cd mbed-simulator
> npm install
> (build-demos.bat ... all demo apps fail to build libmbed ... Windows Error 206 ... filename too long)
> node server.js
> Open http://localhost:7829 in your browser.

The problem system is showing python 2.7.14 for win32 is installed.  It was likely installed prior to Mbed-cli.  I will try your suggestions and report back.

woobagooba, 2019-04-18:
What is the explanation of this issue closure?

linlingao, 2019-04-18:
This was accidentally closed by a script. You can reopen it.

woobagooba, 2019-04-18:
An ARM bot "accidentally" closed this issue without explanation, but I am invited to re-open it?  It is not apparent how to accomplish that.

Your internal Jira is also closed.  Can I re-open that?  

Or will this issue just be dropped on the floor?

bridadan, 2019-04-18:
Sorry about that @woobagooba, reopenning now.

woobagooba, 2019-04-26:
Jan has fixed the Mbed OS Simulator issues. 

How goes the fix for Mbed CLI windows installation?

oclyke, 2019-08-29:
I am experiencing the same issue.
  Windows 10 ver 1803
  Intel 64-bit

I downloaded and ran the installer here https://github.com/ARMmbed/mbed-cli-windows-installer/releases/tag/v0.4.10 and left all options at their default, despite already having Python (both 2.7 and 3.x) and Git installed. 

I can successfully run ```mbed.py``` manually (getting the help menu w/o arguments) using either Python or Python3. 

[image]