ARMmbed/mbed-cli #885: mbed-cli ignores profile

maxgerhardt, 2019-04-11:
I'm in the situation where I must use a custom `profiles/physec_build_profile.json` for my project due to adding a certain C++ language version and other compiler flags. 

Compiling using 

```
mbed compile --profile profiles/physec_build_release_profile.json -m NUCLEO_L476RG -t GCC_ARM
```

works fine. However, when exporting CMake files for later integration with CLion, `mbed-cli` does not respect the `--profile` option.

```
mbed export --profile profiles/physec_build_profile.json -i cmake_gcc_arm -m NUCLEO_L476RG
```

Will produce a `CMakeLists.txt` file which invokes `mbed-cli` with the wrong pfile.

```cmake
##########################################################################
# mbed-cli specific targets
##########################################################################

# detect the build type and select the corresponding cli profile
SET(MBED_BUILD_PROFILE "")
STRING(TOLOWER ${CMAKE_BUILD_TYPE} LOWERCASE_CMAKE_BUILD_TYPE)
IF(LOWERCASE_CMAKE_BUILD_TYPE MATCHES debug)
    SET(MBED_BUILD_PROFILE "mbed-os/tools/profiles/debug.json")
ELSEIF(LOWERCASE_CMAKE_BUILD_TYPE MATCHES relwithdebinfo)
    SET(MBED_BUILD_PROFILE "mbed-os/tools/profiles/develop.json")
ELSEIF(LOWERCASE_CMAKE_BUILD_TYPE MATCHES release)
    SET(MBED_BUILD_PROFILE "mbed-os/tools/profiles/release.json")
ELSEIF(LOWERCASE_CMAKE_BUILD_TYPE MATCHES minsizerel)
    SET(MBED_BUILD_PROFILE "mbed-os/tools/profiles/release.json")
ELSE()
  MESSAGE(WARNING "Build type '${CMAKE_BUILD_TYPE}' is unknown, using debug profile")
  SET(MBED_BUILD_PROFILE "mbed-os/tools/profiles/debug.json")
ENDIF()

# optional custom target to build via mbed-cli
ADD_CUSTOM_TARGET(mbed-cli-build
        COMMAND ${CMAKE_COMMAND} -E echo "mbed compile --build BUILD/${CMAKE_BUILD_TYPE} --profile ${MBED_BUILD_PROFILE}"
        COMMAND mbed compile --build BUILD/${CMAKE_BUILD_TYPE} --profile ${MBED_BUILD_PROFILE}
        WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}
        SOURCES ${SOURCE_FILES} ${SYS_SOURCE_FILES})
``` 

You can see that it first sets `MBED_BUILD_PROFILE` to an empty string and then tries find the correct `MBED_BUILD_PROFILE` from a pre-set list. None of them are the one I specified. 

Subsequently it invokes it when with the `debug.json` profile and fails.

```
/home/max/Downloads/clion-2017.3/bin/cmake/linux/bin/cmake --build /home/max/Documents/my_project/cmake-build-debug --target mbed-cli-build -- -j 2
mbed compile --build BUILD/Debug --profile mbed-os/tools/profiles/debug.json
```

What is the correct way of invoking `mbed-cli` to generate a `CMakeLists.txt` which has the right `--profile`?

bridadan, 2019-04-11:
@maxgerhardt you're invoking the command the right way. I believe support for this was added first in version 1.9.0 of Mbed CLI, what version are you using? (`mbed --version`).

maxgerhardt, 2019-04-11:
Versions as follows (latest pip version)

```
$ mbed --version
1.10.0
$ mbed-cli --version
1.10.0
```

theotherjimmy, 2019-04-11:
@maxgerhardt Could you also include your Mbed OS version?

maxgerhardt, 2019-04-11:
Latest 5.12.1 version. But this bug should be in the `CMakeLists.txt` template renderer and not the mbed-os version.

theotherjimmy, 2019-04-11:
@maxgerhardt The CMakeList.txt template renderer is in Mbed OS under `tools/export/cmake`

maxgerhardt, 2019-04-11:
Well, looking at the hardcoded text in https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.12/tools/export/cmake/CMakeLists.txt.tmpl makes it seem like it is an mbed-os problem then. Should I move the issue?