ARMmbed/mbed-cli #891: Question about 'pip list -l'

andrewc-arm, 2019-04-26:
Hi, @bridadan 

Every time, when I run mbed-cli there is ending printouts of 'pip list -l' like following.
```
[mbed-21318] Exec "/usr/bin/python -m pip list -l" in "/home/andcho01/code/mbed-os-example-ping"
absl-py (0.7.0)
appdirs (1.4.3)
asn1ate (0.6.0)
asn1crypto (0.24.0)
```

It's pushing my compile errors and I have to scroll up to find out my compile errors when I am developing. Could you point to me where I can disable this temporarily when I am developing?

Thanks,

bridadan, 2019-04-26:
Hi @andrewc-arm. It looks like you're running Mbed CLI in "very verbose" mode (`-vv`). This is generally meant for debugging so it dumps a lot of info. You can run it in just "verbose" mode (`-v`) if you still want some extra output, or you can drop the `-v` argument altogether to get just the normal output.

andrewc-arm, 2019-04-27:
Thanks for the quick feedback!

closed 2019-04-27 by andrewc-arm