ARMmbed/mbed-cli #892: Crash on "Auto-installing missing Python modules" since today

amq, 2019-05-03:
Auto-installing missing Python modules fails consistently on CI since today. Interestingly, none of the versions (python, pip, mbed-cli) changed since yesterday. This is not a an OOM-kill issue.

OS:
```
debian:stretch-slim
linux 4.4.0-146-generic
python 2.7.13-2
python-pip 9.0.1-2+deb9u1
```

Commands:
```
pip install mbed-cli
git clone project && cd project
mbed update
```

Log:
```
Collecting mbed-cli
  Downloading https://files.pythonhosted.org/packages/f8/2d/a40a9ef436094dd6fcdf05fd010446baa5c336d36f97ff50415208f31f18/mbed-cli-1.10.0.tar.gz (52kB)
Building wheels for collected packages: mbed-cli
  Running setup.py bdist_wheel for mbed-cli: started
  Running setup.py bdist_wheel for mbed-cli: finished with status 'done'
  Stored in directory: /root/.cache/pip/wheels/1e/69/7c/80a72c06d1f064cbacb8729df2315629e71dfafa1d922c3222
Successfully built mbed-cli
Installing collected packages: mbed-cli
Successfully installed mbed-cli-1.10.0

merge: origin/redacted - not something we can merge
[mbed] ERROR: Missing Python modules were not auto-installed.
       The Mbed OS tools in this program require the following Python modules: colorama, pyserial, prettytable, jinja2, intelhex, junit_xml, pyyaml, requests, mbed_ls, mbed_host_tests, mbed_greentea, beautifulsoup4, fuzzywuzzy, pyelftools, jsonschema, future
       You can install all missing modules by running "pip install -r requirements.txt" in "/redacted/mbed-os"
       On Posix systems (Linux, etc) you might have to switch to superuser account or use "sudo"
---
[mbed] Working path "/redacted" (program)
[mbed] Updating program "redacted" to latest revision in the current branch
[mbed] Adding library ...
[mbed] Adding library ...
[mbed] Auto-installing missing Python modules (colorama, pyserial, prettytable, jinja2, intelhex, junit_xml, pyyaml, requests, mbed_ls, mbed_host_tests, mbed_greentea, beautifulsoup4, fuzzywuzzy, pyelftools, jsonschema, future)...

# immediate crash with exit code 1
```

Strace right before the crash:
```
stat("/redacted/mbed-os", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
stat("/redacted/mbed-os/tools", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
stat("/redacted/mbed-os/tools/requirements.txt", 0x7ffdc73edcd0) = -1 ENOENT (No such file or directory)
stat("/redacted/mbed-os", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
stat("/redacted/mbed-os/requirements.txt", {st_mode=S_IFREG|0644, st_size=247, ...}) = 0
open("/redacted/mbed-os/requirements.txt", O_RDONLY) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=247, ...}) = 0
pipe([4, 5])                            = 0
fcntl(4, F_GETFD)                       = 0
fcntl(4, F_SETFD, FD_CLOEXEC)           = 0
fcntl(5, F_GETFD)                       = 0
fcntl(5, F_SETFD, FD_CLOEXEC)           = 0
pipe([6, 7])                            = 0
fcntl(6, F_GETFD)                       = 0
fcntl(6, F_SETFD, FD_CLOEXEC)           = 0
fcntl(7, F_GETFD)                       = 0
fcntl(7, F_SETFD, FD_CLOEXEC)           = 0
pipe([8, 10])                           = 0
fcntl(8, F_GETFD)                       = 0
fcntl(8, F_SETFD, FD_CLOEXEC)           = 0
fcntl(10, F_GETFD)                      = 0
fcntl(10, F_SETFD, FD_CLOEXEC)          = 0
clone(child_stack=0, flags=CLONE_CHILD_CLEARTID|CLONE_CHILD_SETTID|SIGCHLD, child_tidptr=0x7f14789b3890) = 5929
close(10)                               = 0
mmap(NULL, 1052672, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f147634f000
read(8, "", 1048576)                    = 0
mremap(0x7f147634f000, 1052672, 4096, MREMAP_MAYMOVE) = 0x7f147634f000
munmap(0x7f147634f000, 4096)            = 0
close(5)                                = 0
close(7)                                = 0
close(8)                                = 0
fstat(4, {st_mode=S_IFIFO|0600, st_size=0, ...}) = 0
fcntl(4, F_GETFL)                       = 0 (flags O_RDONLY)
fstat(6, {st_mode=S_IFIFO|0600, st_size=0, ...}) = 0
fcntl(6, F_GETFL)                       = 0 (flags O_RDONLY)
poll([{fd=4, events=POLLIN|POLLPRI}, {fd=6, events=POLLIN|POLLPRI}], 2, -1) = 1 ([{fd=6, revents=POLLIN}])
read(6, "DEPRECATION: The default format "..., 4096) = 213
poll([{fd=4, events=POLLIN|POLLPRI}, {fd=6, events=POLLIN|POLLPRI}], 2, -1) = 1 ([{fd=4, revents=POLLIN}])
read(4, "mbed-cli (1.10.0)\n", 4096)    = 18
poll([{fd=4, events=POLLIN|POLLPRI}, {fd=6, events=POLLIN|POLLPRI}], 2, -1) = 2 ([{fd=4, revents=POLLHUP}, {fd=6, revents=POLLHUP}])
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=5929, si_uid=0, si_status=0, si_utime=50, si_stime=6} ---
close(4)                                = 0
close(6)                                = 0
wait4(5929, [{WIFEXITED(s) && WEXITSTATUS(s) == 0}], 0, NULL) = 5929
fstat(3, {st_mode=S_IFREG|0644, st_size=247, ...}) = 0
lseek(3, 0, SEEK_CUR)                   = 0
lseek(3, 0, SEEK_CUR)                   = 0
fstat(3, {st_mode=S_IFREG|0644, st_size=247, ...}) = 0
read(3, "colorama>=0.3.3\nPySerial>=2.7\nPr"..., 4096) = 247
read(3, "", 4096)                       = 0
pipe([4, 5])                            = 0
fcntl(4, F_GETFD)                       = 0
fcntl(4, F_SETFD, FD_CLOEXEC)           = 0
fcntl(5, F_GETFD)                       = 0
fcntl(5, F_SETFD, FD_CLOEXEC)           = 0
pipe([6, 7])                            = 0
fcntl(6, F_GETFD)                       = 0
fcntl(6, F_SETFD, FD_CLOEXEC)           = 0
fcntl(7, F_GETFD)                       = 0
fcntl(7, F_SETFD, FD_CLOEXEC)           = 0
pipe([8, 10])                           = 0
fcntl(8, F_GETFD)                       = 0
fcntl(8, F_SETFD, FD_CLOEXEC)           = 0
fcntl(10, F_GETFD)                      = 0
fcntl(10, F_SETFD, FD_CLOEXEC)          = 0
clone(child_stack=0, flags=CLONE_CHILD_CLEARTID|CLONE_CHILD_SETTID|SIGCHLD, child_tidptr=0x7f14789b3890) = 5932
close(10)                               = 0
mmap(NULL, 1052672, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f147634f000
read(8, "", 1048576)                    = 0
mremap(0x7f147634f000, 1052672, 4096, MREMAP_MAYMOVE) = 0x7f147634f000
munmap(0x7f147634f000, 4096)            = 0
close(5)                                = 0
close(7)                                = 0
close(8)                                = 0
fstat(4, {st_mode=S_IFIFO|0600, st_size=0, ...}) = 0
fcntl(4, F_GETFL)                       = 0 (flags O_RDONLY)
fstat(6, {st_mode=S_IFIFO|0600, st_size=0, ...}) = 0
fcntl(6, F_GETFL)                       = 0 (flags O_RDONLY)
poll([{fd=4, events=POLLIN|POLLPRI}, {fd=6, events=POLLIN|POLLPRI}], 2, -1) = 1 ([{fd=6, revents=POLLIN}])
read(6, "Command \"python setup.py egg_inf"..., 4096) = 100
poll([{fd=4, events=POLLIN|POLLPRI}, {fd=6, events=POLLIN|POLLPRI}], 2, -1) = 2 ([{fd=4, revents=POLLHUP}, {fd=6, revents=POLLHUP}])
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=5932, si_uid=0, si_status=1, si_utime=277, si_stime=48} ---
close(4)                                = 0
close(6)                                = 0
wait4(5932, [{WIFEXITED(s) && WEXITSTATUS(s) == 1}], 0, NULL) = 5932
close(3)                                = 0
uname({sysname="Linux", nodename="cf11d6cb41ff", ...}) = 0
pipe2([3, 4], O_CLOEXEC)                = 0
clone(child_stack=0, flags=CLONE_CHILD_CLEARTID|CLONE_CHILD_SETTID|SIGCHLD, child_tidptr=0x7f14789b3890) = 5964
close(4)                                = 0
fcntl(3, F_SETFD, 0)                    = 0
fstat(3, {st_mode=S_IFIFO|0600, st_size=0, ...}) = 0
fstat(3, {st_mode=S_IFIFO|0600, st_size=0, ...}) = 0
lseek(3, 0, SEEK_CUR)                   = -1 ESPIPE (Illegal seek)
fstat(3, {st_mode=S_IFIFO|0600, st_size=0, ...}) = 0
read(3, "unknown\n", 4096)              = 8
fstat(3, {st_mode=S_IFIFO|0600, st_size=0, ...}) = 0
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=5964, si_uid=0, si_status=0, si_utime=0, si_stime=0} ---
lseek(3, 0, SEEK_CUR)                   = -1 ESPIPE (Illegal seek)
read(3, "", 4096)                       = 0
close(3)                                = 0
wait4(5964, [{WIFEXITED(s) && WEXITSTATUS(s) == 0}], 0, NULL) = 5964
write(2, "[mbed] ERROR: Missing Python mod"..., 62) = 62
write(2, "       The Mbed OS tools in this"..., 259) = 259
write(2, "       You can install all missi"..., 141) = 141
write(2, "       On Posix systems (Linux, "..., 97) = 97
write(2, "---\n", 4)                    = 4
write(1, "[mbed] Working path \"/redacted/"..., 3233) = 3233
rt_sigaction(SIGINT, {SIG_DFL, [], SA_RESTORER, 0x7f147858b0e0}, {0x55cd56d86660, [], SA_RESTORER, 0x7f147858b0e0}, 8) = 0
close(9)                                = 0
futex(0x7f1477401a2c, FUTEX_WAKE_PRIVATE, 2147483647) = 0
exit_group(1)                           = ?
+++ exited with 1 +++
```

bridadan, 2019-05-03:
@amq It's possible this is related to the problematic release we made yesterday to a subdependency. This should be fixed as of now. If you still experience issues please let me know, thanks!

amq, 2019-05-03:
@bridadan it's working now, thanks!

closed 2019-05-03 by amq