ARMmbed/mbed-cli #893: This variant of Arm Compiler is licensed only for use with Mbed Studio

hierophect, 2019-05-06:
I just got this error after a half hour of installations and messing with environmental variables trying to get ARM6 to work with mbed cli. 
```
Building project mbed-os-program (NUCLEO_F103RB, ARMC6)
Scan: mbed-os-program
Compile [  0.1%]: mbed_tz_context.c
Compile [  0.2%]: except.S
[Error] @0,0: A9940E: This variant of Arm Compiler is licensed only for use with Mbed Studio
[ERROR] Error: A9940E: This variant of Arm Compiler is licensed only for use with Mbed Studio
Product: ARM Compiler 6.11 for Mbed Studio
Component: ARM Compiler 6.11
Tool: armasm [5d3b3500]
```

If Mbed Studio's arm compiler is indeed not licensed for mbed-cli use, maybe there shouldn't be a huge section on the CLI website talking about how to set it up?:
https://os.mbed.com/docs/mbed-os/v5.12/tools/after-installation-configuring-mbed-cli.html

bridadan, 2019-05-06:
Hi @hierophect, sorry to hear about your installation troubles, perhaps I can help. From the output it looks like you've definitely completed step 2 from the link you posted (otherwise you wouldn't be getting that error). Can you answer the following questions for me?

- What OS are you using?
    - If you are using Windows, are you using the normal command shell (cmd.exe) or are you using something else? Examples include Git Bash, Windows Subsytem for Linux, cygwin, etc
- How did you set the environment variable `ARMLMD_LICENSE_FILE`? You should be able to run the following command from the terminal you're using to run Mbed CLI and see the value you set:
    - On Mac: `echo $ARMLMD_LICENSE_FILE`
    - On Windows (using cmd.exe): `echo %ARMLMD_LICENSE_FILE%`

hierophect, 2019-05-06:
I'm on windows via git bash. 

I'm using the following file, as specified in the setup instructions. I set it in windows advanced settings. I can verify it's part of my environment via the usual non-mbed commands. 
C:\Users\<my username>\AppData\Local\Mbed Studio\mbed-studio-tools\ac6-license.dat

bridadan, 2019-05-07:
@hierophect what version of Mbed OS are you using? I'm realizing that docs page could be more clear that using Mbed Studio's version of ARMC6 is only supported in 5.12.0 and later.

If you are using 5.12.0, could you post the verbose output? `mbed compile -v`.

bridadan, 2019-05-07:
I've been able to reproduce the issue, thanks for the details. After doing a bisect it looks like this PR broke it: https://github.com/ARMmbed/mbed-os/pull/10182.

I have a fix up here: https://github.com/ARMmbed/mbed-os/pull/10539

linlingao, 2019-05-09:
The reported issue has been recreated and fix has been provided, closing.

closed 2019-05-10 by ciarmcom

lazyoracle, 2020-06-19:
Has this changed in the recent versions of mbed-cli and Mbed Studio? I am currently using the latest stable release of mbed-cli `1.10.4` and Mbed Studio `1.0.0` for mbed-os `6.0`. 

**How do I direct `mbed` to use the `ARMC6` compiler that comes preinstalled with Mbed Studio?** 

The docs regarding this seem to have changed from [v5.12](https://os.mbed.com/docs/mbed-os/v5.12/tools/after-installation-configuring-mbed-cli.html) to [v6.0](https://os.mbed.com/docs/mbed-os/v6.0/build-tools/install-and-set-up.html).