ARMmbed/mbed-cli #894: Building mbed-os-example-blinky compiles all mbed os?

microbit-carlos, 2019-05-14:
Building the default simple blinky results in what is probably all of mbedOS being compiled, taking about hour to finish, and stopping half way through.

To replicate, all I have done is:

```
$ mbed import mbed-os-example-blinky
$ cd mbed-os-example-blinky/
$ mbed compile -m K64F -t GCC_ARM
```

Expandable output:

<details>
  <summary>$ mbed compile -m K64F -t GCC_ARM</summary>
<pre>

[mbed] Working path "/home/vagrant/vagrant_shared/deleteme/mbed-os-example-blinky" (program)
[Warning] @,: Compiler version mismatch: Have 7.3.1; expected version >= 6.0.0 and < 7.0.0
Building project mbed-os-example-blinky (K64F, GCC_ARM)
Scan: mbed-os-example-blinky
Compile [  0.1%]: except.S
Compile [  0.2%]: main.cpp
Compile [  0.4%]: mbed_tz_context.c
Compile [  0.5%]: mbed_fault_handler.c
Compile [  0.6%]: NanostackRfPhyAtmel.cpp
Compile [  0.7%]: MCR20Drv.c
Compile [  0.8%]: NanostackRfPhyMcr20a.cpp
Compile [  0.9%]: at24mac.cpp
Compile [  1.1%]: rf_configuration.c
Compile [  1.2%]: NanostackRfPhys2lp.cpp
Compile [  1.3%]: psa_attest_inject_key.c
Compile [  1.4%]: attest_boot_status_loader.c
Compile [  1.5%]: psa_initial_attestation_api.c
Compile [  1.6%]: attest_crypto.c
Compile [  1.8%]: attest_crypto_keys.c
Compile [  1.9%]: attest_iat_claims_loader.c
[Warning] attest_iat_claims_loader.c@91,30: implicit declaration of function 'malloc' [-Wimplicit-function-declaration]
[Warning] attest_iat_claims_loader.c@124,9: implicit declaration of function 'free' [-Wimplicit-function-declaration]
Compile [  2.0%]: attestation_bootloader_data.c
Compile [  2.1%]: psa_attestation_stubs.c
Compile [  2.2%]: attest_token.c
Compile [  2.3%]: psa_inject_attestation_key_impl.c
Compile [  2.5%]: t_cose_sign1_sign.c
Compile [  2.6%]: attestation_core.c
[Warning] attestation_core.c@719,28: comparison is always false due to limited range of data type [-Wtype-limits]
[Warning] attestation_core.c@844,13: variable 'alg_select' set but not used [-Wunused-but-set-variable]
Compile [  2.7%]: t_cose_util.c
Compile [  2.8%]: ieee754.c
Compile [  2.9%]: UsefulBuf.c
Compile [  3.0%]: qcbor_decode.c
Compile [  3.2%]: qcbor_encode.c
Compile [  3.3%]: UsefulBuf_Tests.c
Compile [  3.4%]: half_to_double_from_rfc7049.c
Compile [  3.5%]: qcbor_decode_tests.c
Compile [  3.6%]: float_tests.c
Compile [  3.8%]: qcbor_encode_tests.c
Compile [  3.9%]: platform_emul.c
Compile [  4.0%]: platform_srv_impl.c
Compile [  4.1%]: run_tests.c
Compile [  4.2%]: psa_storage_common_impl.cpp
Compile [  4.3%]: psa_prot_internal_storage.cpp
Compile [  4.5%]: protected_storage.cpp
Compile [  4.6%]: pits_impl.cpp
Compile [  4.7%]: fslittle_test.c
Compile [  4.8%]: SDBlockDevice.cpp
Compile [  4.9%]: FlashIAPBlockDevice.cpp
Compile [  5.0%]: ESP8266Interface.cpp
Compile [  5.2%]: ESP8266.cpp
Compile [  5.3%]: AnalogIn.cpp
Compile [  5.4%]: BusIn.cpp
[... 865 lines cut ...]
| Module             |         .text |       .data |        .bss |
|--------------------|---------------|-------------|-------------|
| [fill]             |     150(+150) |       4(+4) | 2113(+2113) |
| [lib]/c.a          | 30675(+30675) | 2472(+2472) |     89(+89) |
| [lib]/gcc.a        |   3112(+3112) |       0(+0) |       0(+0) |
| [lib]/misc         |     204(+204) |       4(+4) |     28(+28) |
| [lib]/nosys.a      |       32(+32) |       0(+0) |       0(+0) |
| main.o             |   1062(+1062) |       0(+0) |     12(+12) |
| mbed-os/cmsis      |   1033(+1033) |       0(+0) |       0(+0) |
| mbed-os/components |     149(+149) |       0(+0) |       0(+0) |
| mbed-os/drivers    |       74(+74) |       0(+0) |       0(+0) |
| mbed-os/features   |       42(+42) |       0(+0) |   184(+184) |
| mbed-os/hal        |   1695(+1695) |       8(+8) |   130(+130) |
| mbed-os/platform   |   4988(+4988) |   260(+260) |   257(+257) |
| mbed-os/rtos       |   8571(+8571) |   168(+168) | 5973(+5973) |
| mbed-os/targets    |   9585(+9585) |     12(+12) |   382(+382) |
| Subtotals          | 61372(+61372) | 2928(+2928) | 9168(+9168) |
Total Static RAM memory (data + bss): 12096(+12096) bytes
Total Flash memory (text + data): 64300(+64300) bytes

Image: ./BUILD/K64F/GCC_ARM/mbed-os-example-blinky.bin
</pre>
</details>

Total compilation took 55min in a VM (allocated and fully utilising 2 i7 cores @ 4Ghz, 3K RAM) running Ubuntu 16.04, GCC ARM v7.3.1, mbed-cli v1.10.0 (on Python 2.7.12).

cederom, 2019-05-14:
Hey Carlos! Use the `-j` switch to perform a parallel build that speeds up the build noticably. I use `mbed compile -j8 —color` that performs parallel build using 8 threads and colors the build output for better Warning and Error visibility (I wish it gets default).

That does not solve the full build problem but relieves the painful time a bit :-)

microbit-carlos, 2019-05-14:
Thank you @cederom, although using different `-j` values doesn't seem to noticeably affect the build speed, could it be that the problem is the compilation speed?

It is the first time I am using mbed-cli, so I don't really know what it is meant to do on a first build, is it normal to compile all the files listed in the two outputs I added to the issue description?

cederom, 2019-05-14:
Yes the first time whole MBED needs to be build because those binaries are then linked with your portion of code in order to create a working firmware image.

The `-j1` `-j2` and `-j4` should show the difference, depending on your CPU it should use more cores to compile in parallel, around 2x times.. more threads does not necessary will do any difference except glitching the system performance. I just use `-j8` to suck all from my system :-)

The from-scratch build setup looks something like this, please take a look if you have not missed anything:
1. First you need to prepare the valid build environment on your build platform: Python (VirtualEnv), git+mercurial, ARM_GCC.
2. Get Python3.7.
3. Go to project top directory and setup Python VirtualEnv (it makes local copy of python environment that does not affect your system python): `python3.7 -m venv <venv37_directory_location>` (lets say `../venv37`).
4. Activate Python VirtualEnv: `source ven37/bin/activate` (this will put you into local python environment).
5. Install mbed tools: `pip install -U pip mbed-cli mbed-ls pyocd mercurial`.
6. Download/update mbed-os: `mbed update`.
7. Install mbed-os dependencies: `pip install -U mbed-os/requirements.txt`.
8. In case you did not install ARM_GCC alreaty take one from https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm and add its binary to system path so its avaialbe to mbed.
9. Now you are ready to go with `mbed compile` :-)

Some remarks:
* do not mofiy anything inside `mbed-os`.
* make sure your `mbed-os` is in sync with the version specified in `mbed-os.lib` file. Update to latest (5.12.3 if necessary).
* You can see what is the exact build problem with `-v` switch to `mbed compile` it will show full compiler output and messages.
* You can set `mbed target K64F` so you dont have to write it each time.
* You can set `mbed toolchain GCC_ARM` so you hont have to write it each time.

Here is the documentation on environment setup: https://os.mbed.com/docs/mbed-os/v5.6/tools/setup.html

microbit-carlos, 2019-05-14:
Thanks @cederom!

> Yes the first time whole MBED needs to be build because those binaries are then linked with your portion of code in order to create a working firmware image. 

I've downloaded Mbed Studio locally and I can see a similar output, so I guess it does compile through all those file even though it only needs to link a tiny portion of them.

Looks like this is by design, is there a way to only compile what it needs?

With Mbed studio in macOS the main difference (apart from using Arm Compiler instead of GCC ARM) is that instead of 55min it takes <1min, although it is using the 4 physical cores + 4 virtual ones from hyperthreading, instead of the 2 cores allocated by VirtualBox.

>The -j1 -j2 and -j4 should show the difference, depending on your CPU it should use more cores to compile in parallel, around 2x times.. more threads does not necessary will do any difference except glitching the system performance. I just use -j8 to suck all from my system :-)

Using just `mbed compile` on its own in the VM with 2 allocated cores and I can see always two GCC ARM threads automatically maxing out both cores, and running it on the host OS (4 physical, 4 virtual) and I can see about 8 threads. So I assume something automatically figures out how many threads to spawn.

On the same VM with 2 cores yotta with GGC ARM builds projects in just a few minutes.

In case it made any difference I also tried using the tooling with Python 3. To do that I've follow these steps (excluding `pip install mercurial` from your instructions, instead mercurial was already installed via apt-get):

```
$ sudo apt-get install python3-pip python3-venv

$ pip3 install --user virtualenv

$ python3 -m venv env

$ source env/bin/activate

(env) $ pip install -U pip mbed-cli mbed-ls pyocd

(env) $ which mbed
/home/vagrant/vagrant_shared/deleteme/env/bin/mbed

(env) $ mbed config -G GCC_ARM_PATH "$(dirname "$(which arm-none-eabi-gcc)")"
[mbed] /usr/bin now set as global GCC_ARM_PATH

(env) $ mbed import mbed-os-example-blinky
[mbed] Working path "/home/vagrant/vagrant_shared/deleteme" (directory)
[mbed] Importing program "mbed-os-example-blinky" from "https://github.com/ARMmbed/mbed-os-example-blinky" at latest revision in the current branch
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #0063e5de32fc

(env) $ cd mbed-os-example-blinky/

(env) $ mbed update
[mbed] Working path "/home/vagrant/vagrant_shared/deleteme/mbed-os-example-blinky" (program)
[mbed] Updating program "mbed-os-example-blinky" to latest revision in the current branch
[mbed] Updating library "mbed-os" to rev #0063e5de32fc (tag: mbed-os-5.12.1)

(env) $ mbed compile -m K64F -t GCC_ARM -j4 --color
``` 

And the compilation speed was the same, after 10min it was still at about 15%.

Following the same instructions on my host OS (macOS), and using `-j2` and it compiled in about 1min30s.

> Compilaion inside VM may be slowe than on the host machine. Please try to setup you local build environment and verify. Performance may be highly affected by VM Hypervisor. For instance I have noticed dramatic peformance degradation in VirtualBox6 on macOS 10.14 barely usable at all (my FreeBSD VBox5 works many times faster on a definitely slower host).

2min vs 55min is too much of a difference. Do you have any recommendation to try to figure out the problem?

cederom, 2019-05-14:
The MBED structure and build questions should go to the MBED team that I am not a part of :-)

Try installing the MBED in local environment on the host machine, not inside VM, I guess the problem is caused by VM itself. On both macOS and FreeBSD full mbed build takes a few minutes for me. If build on host works fine then we would confirm the problem is within the VM (maybe some tuning like enabling hardware VM acceleration will help).

linlingao, 2019-05-14:
Building anything in VM is slower than native machine. I don't think this is Mbed specific issue. I'm not sure what we can do here.

closed 2019-05-15 by ciarmcom

microbit-carlos, 2019-05-16:
I understand it will be slower, but in this case mbed-cli is **35 times slower** on the VM, which makes it unusable. In the same VM building with yotta is about 10 slower compared with native, so something feels off. I'll keep looking into the VM configuration, but any tips would be greatly appreciated.