ARMmbed/mbed-cli #902: NUCLEO_F103RB not supported by uvision5

fengshengchang, 2019-06-26:
When i was exporting to uvision, I got an error: NUCLEO_F103RB not supported by uvision5
I don`t not how to solve this problem.  

G:\nbproject>mbed export -i uvision -m NUCLEO_F103RB --source nbapp --source mbed-os -v
[mbed] Working path "G:\nbproject" (program)
[mbed] Exec "d:\python27\python.exe -u G:/nbproject/mbed-os\tools\project.py -i uvision -m NUCLEO_F103RB --profile debug --source nbapp --source mbed-os" in "G:\nbproject"
NUCLEO_F103RB not supported by uvision5
[mbed] ERROR: "d:\python27\python.exe" returned error.
       Code: 2
       Path: "G:\nbproject"
       Command: "d:\python27\python.exe -u G:/nbproject/mbed-os\tools\project.py -i uvision -m NUCLEO_F103RB --profile debug --source nbapp --source mbed-os"
       Tip: You could retry the last command with "-v" flag for verbose output
---

linlingao, 2019-06-28:
Do you have to use uvision5 (armcc), will uvision6 do (armclang)?

This should work:
mbed export -i uvision6 -m nucleo_f103rb

fengshengchang, 2019-06-29:
> 
> 
> Do you have to use uvision5 (armcc), will uvision6 do (armclang)?
> 
> This should work:
> mbed export -i uvision6 -m nucleo_f103rb

closed 2019-06-29 by fengshengchang

fengshengchang, 2019-06-29:
> 
> 
> Do you have to use uvision5 (armcc), will uvision6 do (armclang)?
> 
> This should work:
> mbed export -i uvision6 -m nucleo_f103rb
Hi linlingao
    Thank you very much, I tried your suggestion， It worked.