ARMmbed/mbed-cli #904: Can't install requirements.txt for Linux

maxgerhardt, 2019-07-02:
```
$ mbed --version
1.10.0
```

So, when creating a new project, the following occurs:

```
$ mbed new "test"
[mbed] Working path "/home/max/Documents" (directory)
[mbed] Creating new program "test (git)
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at branch/tag "latest"
[mbed] Updating reference "mbed-os" -> "https://github.com/ARMmbed/mbed-os/#7482462434d5cf718177653ef797547a976a7c5e"
[mbed] Auto-installing missing Python modules (hidapi, pywin32, wmi)...
[mbed] ERROR: Missing Python modules were not auto-installed.
       The Mbed OS tools in this program require the following Python modules: hidapi, pywin32, wmi
       You can install all missing modules by running "pip install -r requirements.txt" in "/home/max/Documents/test/mbed-os"
       On Posix systems (Linux, etc) you might have to switch to superuser account or use "sudo"
---
``` 

So I install the requirements with python2 and python3 to be sure. However, the marked packages are not installed because I'm using a Linux machine

```
sudo pip3 install -U -r FW_Impulsgeber/mbed-os/requirements.txt
The directory '/home/max/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/max/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Ignoring hidapi: markers 'platform_system != "Linux"' don't match your environment
Ignoring pywin32: markers 'platform_system == "Windows"' don't match your environment
Ignoring wmi: markers 'platform_system == "Windows"' don't match your environment
```

However, installing the requirements.txt file will result in `pip` refusing to install it, because I have a Linux machine and these packages are thought for Windows/Mac

```
sudo pip3 install -U -r test/mbed-os/requirements.txt
The directory '/home/max/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/max/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Ignoring hidapi: markers 'platform_system != "Linux"' don't match your environment
Ignoring pywin32: markers 'platform_system == "Windows"' don't match your environment
Ignoring wmi: markers 'platform_system == "Windows"' don't match your environment
```
It does install all other packages though. This is done for python2 and python3. 

Installing these packages has no effect on the error messages. `mbed-cli` still refuses to fully create a new project. The only "fix" is to do it with `mbed new --no-requirements`. 

Can you reproduce this problem under Linux? I'm using Ubuntu 18.10 on an x64 machine here.

madchutney, 2019-07-08:
We believe this issue has been fixed in Mbed CLI master, it requires a new release of the tool which we will be doing shortly https://github.com/ARMmbed/mbed-cli/pull/889

maxgerhardt, 2019-07-10:
Given that the pull request was 3 months ago, when can a new package version of `mbed-cli` be expected? I just did a `pip install -U mbed-cli` and still got `1.10.0`.

madchutney, 2019-07-10:
When it has been tested, which is our current highest priority.

bu5hm4nn, 2019-08-03:
This just broke our CI when upgrading to mbed 5.13. Temporary solution is to build mbed-cli directly from github repo:
```
sudo pip install -U git+https://github.com/ARMmbed/mbed-cli
```

maxgerhardt, 2019-08-13:
> a new release of the tool which we will be doing shortly #889

Must be one hell of testing phase if it takes 2 months to test 11 commits which includes a critical fix.

[image]

madchutney, 2019-08-22:
Thanks for your patience, mbed-cli has been released and should resolve this issue.

closed 2019-08-22 by madchutney