ARMmbed/mbed-cli #905: Misleading AttributeError: '_queue.SimpleQueue' object has no attribute 'queue' when compiler fails

desowin, 2019-07-18:
There is completely misleading error message when underlying command fails. This is easiest to reproduce by adding invalid commandline switch to gcc profile (eg. open mbed-os/tools/profiles/develop.json and add "-invalid-flag" to "common" array) and then building project with GCC.

The example output without verbose flag is:
```
$ mbed compile
Building project mbed-os-example-blinky (DISCO_F746NG, GCC_ARM)
Scan: mbed-os-example-blinky
Compile [  0.5%]: mbed_tz_context.c
[ERROR] '_queue.SimpleQueue' object has no attribute 'queue'
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] ERROR: "c:\python37\python.exe" returned error.
Code: 1
Path: "C:\work\mbed\mbed-os-example-blinky"
Command: "c:\python37\python.exe -u C:\work\mbed\mbed-os-example-blinky\mbed-os\tools\make.py -t GCC_ARM -m DISCO_F746NG --source . --build .\BUILD\DISCO_F746NG\GCC_ARM"
Tip: You could retry the last command with "-v" flag for verbose output
---
[mbed] WARNING: Using Python 3 with Mbed OS 5.8 and earlier can cause errors with compiling, testing and exporting
---
[mbed] Working path "C:\work\mbed\mbed-os-example-blinky" (program)
--
```

This misleading error message is made clear by using `mbed compile -v`
```
[DEBUG] Output: arm-none-eabi-g++: error: unrecognized command line option '-invalid-flag'; did you mean '-Winvalid-pch'?
Traceback (most recent call last):
File "C:\work\mbed\mbed-os-example-blinky\mbed-os\tools\toolchains\mbed_toolchain.py", line 523, in compile_queue
res['command']
File "C:\work\mbed\mbed-os-example-blinky\mbed-os\tools\toolchains\mbed_toolchain.py", line 650, in compile_output
raise ToolException(stderr)
tools.utils.ToolException: arm-none-eabi-g++: error: unrecognized command line option '-invalid-flag'; did you mean '-Winvalid-pch'?
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "C:\work\mbed\mbed-os-example-blinky\mbed-os\tools\make.py", line 76, in wrapped_build_project
src_dir, build_dir, mcu, *args, **kwargs
File "C:\work\mbed\mbed-os-example-blinky\mbed-os\tools\build_api.py", line 596, in build_project
objects = toolchain.compile_sources(resources, sorted(resources.get_file_paths(FileType.INC_DIR)))
File "C:\work\mbed\mbed-os-example-blinky\mbed-os\tools\toolchains\mbed_toolchain.py", line 459, in compile_sources
return self.compile_queue(queue, objects)
File "C:\work\mbed\mbed-os-example-blinky\mbed-os\tools\toolchains\mbed_toolchain.py", line 527, in compile_queue
if p._taskqueue.queue:
AttributeError: '_queue.SimpleQueue' object has no attribute 'queue'
```

0xc0170, 2019-08-14:
We had similar issue when there was definition duplicate. Had us confused for a while to realize what broke it. I don't understand why queue attribute failed though. Looks like tools bug.

0xc0170, 2019-09-05:
@desowin Have you figure this out? I am now getting the same error locally for blinky baremetal

```
mbed-os-example-blinky-baremetal (master -> origin)
mbed compile -m NRF52840_DK -t ARM
```

Same for py2 and py3. It's just this target that is failing.

0xc0170, 2019-09-05:
cc @ARMmbed/mbed-os-tools

desowin, 2019-09-05:
> @desowin Have you figure this out?

I am not sure what you mean. I didn't work on fixing this issue. I am not planning on fixing it either.

I have observed the AttributeError in my project (it happened rarely) and then identified how to reproduce the problem. The (in my opinion simplest) way to reproduce the problem is in the issue description.

acabarbaye, 2019-09-05:
I think it happens when you run mbed-cli using Python 3 instead of Python 2.7

acabarbaye, 2019-09-05:
It is not just for this target. I get the same issue for quite a few targets: And the error is a Python error due to this line: https://github.com/ARMmbed/mbed-os/blob/f06145db13c38f54d9d33b36c8e8859ddd912d9e/tools/toolchains/mbed_toolchain.py#L559

mark-edgeworth, 2019-09-18:
Thank you for the clear explanation, and for providing steps to reproduce this issue. I have tried this on both python2 and python 3; this appears to be a python-3 incompatibility. The codebase is still not completely python-3 compatible I'm afraid. For the moment the workaround is to use a virtual environment with python 2(.7).

mark-edgeworth, 2019-09-18:
Actually it looks like this has been resolved in the next version of mbed-os (5.14.0) which is in release testing now.

SeppoTakalo, 2019-11-01:
Happens in current `master`:

```
Compile [  0.1%]: mbed_tz_context.c
'_queue.SimpleQueue' object has no attribute 'queue'
Failed to build library
Memory map breakdown for built projects (values in Bytes):
| name | target | toolchain | static_ram | total_flash |
|------|--------|-----------|------------|-------------|
Traceback (most recent call last):
  File "/Users/septak01/src/mbed-os/tools/test.py", line 325, in main
    status = print_report_exporter.report(build_report)
  File "/Users/septak01/src/mbed-os/tools/test_exporters.py", line 92, in report
    return self.exporter_print(test_summary_ext, print_log_for_failures=print_log_for_failures)
  File "/Users/septak01/src/mbed-os/tools/test_exporters.py", line 345, in exporter_print
    raise Exception("'test_run' did not have a 'result' value")
Exception: 'test_run' did not have a 'result' value
[ERROR] 'test_run' did not have a 'result' value
[mbed] ERROR: "/Users/septak01/.virtualenvs/mbed/bin/python3.7" returned error.
       Code: 1
       Path: "/Users/septak01/src/mbed-os"
       Command: "/Users/septak01/.virtualenvs/mbed/bin/python3.7 -u /Users/septak01/src/mbed-os/tools/test.py -D MBED_TEST_MODE -t GCC_ARM -m K64F --source . --build ./BUILD/tests/K64F/GCC_ARM --test-spec ./BUILD/tests/K64F/GCC_ARM/test_spec.json --build-data ./BUILD/tests/K64F/GCC_ARM/build_data.json -n tests-netsocket* --greentea"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

How to resolve this?

SeppoTakalo, 2019-11-01:
Ah.. the problem was that I have ccache setup, but the compiler was not in the PATH.
Well, that is an easy way to replicate this issue.