ARMmbed/mbed-cli #909: mbed deploy doesn't work within a corktree

pan-, 2019-08-20:
I tried to use `mbed deploy` to deploy `.lib` file and it appears that it doesn't work within a git worktree. What can I do to work around that limitation ?

mark-edgeworth, 2019-09-11:
Could you provide a little more information as to what is not working for you please? For me, although the code does complain that the program does not use source control (which it does, albeit as a worktree), the deploy seems to do what is expected.

mark-edgeworth, 2019-09-24:
Thanks for this! I've done a little digging on this myself. As you might know, the build tools for mbed OS support both git and mercurial as SCM systems. The tools distinguish between the two by looking for a directory called .git or .hg in the root. When operating a worktree, the .git directory is actually a file, and the tools don't handle this.
Whilst it is simple to fix this single issue, I'm concerned that changing this will bring in other potential problems that are related solely to the worktree concept. I am sorry for any problems that this may bring you. I will keep this issue on our backlog and see what we can do.

adbridge, 2019-11-01:
Apologies, this was closed in error by the automated bot - re-opening.

mark-edgeworth, 2019-12-04:
To be honest we don't have the resource available to tackle this one.

closed 2019-12-04 by mark-edgeworth