ARMmbed/mbed-cli #913: mbed-cli unable to write config file

ssnover95, 2019-08-23:
My current mbed-cli version is v1.10.1. I am trying to specify the default hardware and compiler for my project. In each case there is a warning that it could not write the config file. Then the program contradicts this and claims it was able to. Attempting to compile shows that it could not.

```
ssnover@styrka:~/develop/vl53l0x-test$ mbed --version
1.10.1
ssnover@styrka:~/develop/vl53l0x-test$ mbed toolchain GCC_ARM
[mbed] WARNING: Python 3 is not yet fully supported: Python errors may occur when compiling, testing and exporting
---
[mbed] Working path "/home/ssnover/develop/vl53l0x-test" (program)
[mbed] WARNING: Unable to write config file /home/ssnover/develop/vl53l0x-test/.mbed
---
[mbed] GCC_ARM now set as default toolchain in program "vl53l0x-test"
ssnover@styrka:~/develop/vl53l0x-test$ mbed target DISCO_L475VG_IOT01A
[mbed] WARNING: Python 3 is not yet fully supported: Python errors may occur when compiling, testing and exporting
---
[mbed] Working path "/home/ssnover/develop/vl53l0x-test" (program)
[mbed] WARNING: Unable to write config file /home/ssnover/develop/vl53l0x-test/.mbed
---
[mbed] DISCO_L475VG_IOT01A now set as default target in program "vl53l0x-test"
ssnover@styrka:~/develop/vl53l0x-test$ mbed compile
[mbed] WARNING: Python 3 is not yet fully supported: Python errors may occur when compiling, testing and exporting
---
[mbed] Working path "/home/ssnover/develop/vl53l0x-test" (program)
[mbed] ERROR: No targets were detected.
       Please make sure a target board is connected to this system.
       Alternatively, you can specify target using the -m switch or set default target using command "mbed target"
---

```

Here's the contents of the project's .mbed file:
```
ROOT=.
```

ssnover95, 2019-08-23:
Additional info that might be helpful: the exception appears to be thrown here https://github.com/ARMmbed/mbed-cli/blob/2756f0a3de843da7c877cefd0f4080612a0d18d4/mbed/mbed.py#L1926

And here's my trying to do approximately that in my interpreter:
```
ssnover@styrka:~/develop/vl53l0x-test$ python3
Python 3.6.8 (default, Jan 14 2019, 11:02:34) 
[GCC 8.0.1 20180414 (experimental) [trunk revision 259383]] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import os
>>> fl = os.path.join("/home/ssnover/develop/vl53l0x-test", ".mbed")
>>> with open(fl, 'w') as f:
...     f.write('\n' + "test")
... 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
PermissionError: [Errno 13] Permission denied: '/home/ssnover/develop/vl53l0x-test/.mbed'
>>> 
```

ssnover95, 2019-08-23:
I found (thanks to running the code manually in an interpreter) that running with sudo solves the problem. Is there something wrong with the way I performed my install perhaps?

madchutney, 2019-09-10:
This is a common kind of error seen with permissions, typically the user running the mbed command line does not have access to the directory and/or file which is why sudo helps.

The feedback from mbed-cli is misleading, as the write does indeed fail, thanks for raising the issue.

ssnover95, 2019-12-05:
The reason this is the case is that when I call `mbed new .` in a directory, I get an error with the following output:
```
Traceback (most recent call last):
  File "/home/ssnover/.local/bin/mbed", line 7, in <module>
    from mbed.mbed import main
ModuleNotFoundError: No module named 'mbed'
```

In order for the call to be successful, I need to call it with sudo, which makes the owner of all of the files root:
```
ssnover@styrka:mbed-test$ ls -la
drwxr-xr-x    - root  4 Dec 21:14 .git
.rw-r--r--    7 root  4 Dec 21:14 .mbed
drwxr-xr-x    - root  4 Dec 21:14 mbed-os
.rw-r--r--   77 root  4 Dec 21:14 mbed-os.lib
.rw-r--r--  108 root  4 Dec 21:14 mbed_app.json
.rw-r--r-- 1.3k root  4 Dec 21:14 mbed_settings.py
```

I found that I in fact had two different versions of mbed installed which my normal user and root user referenced separately:
```
ssnover@styrka:mbed-test$ which mbed
/home/ssnover/.local/bin/mbed
ssnover@styrka:mbed-test$ sudo which mbed
[sudo] password for ssnover: 
/usr/local/bin/mbed
```

So I re-installed mbed taking care to not use sudo and found that after calling `mbed new .` worked fine. Hopefully there's enough keyword density in this thread for anyone else who makes the same silly mistake to find it.

closed 2019-12-05 by ssnover95