ARMmbed/mbed-cli #915: Empty program still links a lot of code making a rather heavy .hex

ladislas, 2019-08-26:
Hi there!

Coming from the AVR world, I've recently started playing with ARM and mbed to my great pleasure. My way of exploring this new paradigm is to start from the examples, run them, tear them down until it stops working and build again one brick at a time to understand how things work together.

To start with, I've tried compiling an empty program `int main(void) { return 0; }`, expecting the resulting .hex to be close to empty.

But this is what I got:

```console
$ mbed compile --stats-depth=10 --color -j8 --profile release

[mbed] Working path "/Users/ladislas/dev/leka/LKAlphaOS-mbed" (program)
[Warning] @,: Compiler version mismatch: Have 8.2.1; expected version >= 6.0.0 and < 7.0.0
Building project LKAlphaOS-mbed (NRF52_DK, GCC_ARM)
Scan: LKAlphaOS-mbed
Link: LKAlphaOS-mbed
Elf2Bin: LKAlphaOS-mbed
| Module                                                                                                                  |     .text |    .data |    .bss |
|-------------------------------------------------------------------------------------------------------------------------|-----------|----------|---------|
| [fill]                                                                                                                  |    30(+0) |    4(+0) |  26(+0) |
| [lib]/c.a/lib_a-callocr.o                                                                                               |    96(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-ctype_.o                                                                                                |     0(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-dtoa.o                                                                                                  |  3952(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-errno.o                                                                                                 |    12(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-freer.o                                                                                                 |   652(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-impure.o                                                                                                |     0(+0) | 1068(+0) |   0(+0) |
| [lib]/c.a/lib_a-init.o                                                                                                  |    72(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-locale.o                                                                                                |     0(+0) |  364(+0) |   0(+0) |
| [lib]/c.a/lib_a-localeconv.o                                                                                            |     8(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-malloc.o                                                                                                |    16(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-mallocr.o                                                                                               |  1392(+0) | 1040(+0) |  52(+0) |
| [lib]/c.a/lib_a-mbtowc_r.o                                                                                              |    44(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-memchr.o                                                                                                |   160(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-memcpy.o                                                                                                |   308(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-memmove.o                                                                                               |   208(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-memset.o                                                                                                |   160(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-mprec.o                                                                                                 |  1548(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-reallocr.o                                                                                              |   876(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-reent.o                                                                                                 |     0(+0) |    0(+0) |   4(+0) |
| [lib]/c.a/lib_a-s_frexp.o                                                                                               |   112(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-sbrkr.o                                                                                                 |    36(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-strlen.o                                                                                                |   220(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-strncpy.o                                                                                               |   116(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-svfiprintf.o                                                                                            |   248(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-svfprintf.o                                                                                             |  6724(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-vsnprintf.o                                                                                             |   136(+0) |    0(+0) |   0(+0) |
| [lib]/c.a/lib_a-wctomb_r.o                                                                                              |    28(+0) |    0(+0) |   0(+0) |
| [lib]/gcc.a/_aeabi_uldivmod.o                                                                                           |    48(+0) |    0(+0) |   0(+0) |
| [lib]/gcc.a/_arm_addsubdf3.o                                                                                            |   888(+0) |    0(+0) |   0(+0) |
| [lib]/gcc.a/_arm_cmpdf2.o                                                                                               |   272(+0) |    0(+0) |   0(+0) |
| [lib]/gcc.a/_arm_fixdfsi.o                                                                                              |    80(+0) |    0(+0) |   0(+0) |
| [lib]/gcc.a/_arm_muldivdf3.o                                                                                            |  1060(+0) |    0(+0) |   0(+0) |
| [lib]/gcc.a/_arm_unorddf2.o                                                                                             |    44(+0) |    0(+0) |   0(+0) |
| [lib]/gcc.a/_dvmd_tls.o                                                                                                 |     4(+0) |    0(+0) |   0(+0) |
| [lib]/gcc.a/_udivmoddi4.o                                                                                               |   720(+0) |    0(+0) |   0(+0) |
| [lib]/misc/crt0.o                                                                                                       |   116(+0) |    0(+0) |   0(+0) |
| [lib]/misc/crtbegin.o                                                                                                   |    64(+0) |    4(+0) |  28(+0) |
| [lib]/misc/crtend.o                                                                                                     |     0(+0) |    0(+0) |   0(+0) |
| [lib]/misc/crti.o                                                                                                       |     0(+0) |    0(+0) |   0(+0) |
| [lib]/misc/crtn.o                                                                                                       |     0(+0) |    0(+0) |   0(+0) |
| mbed-os/drivers/Serial.o                                                                                                |    54(+0) |    0(+0) |   0(+0) |
| mbed-os/drivers/UARTSerial.o                                                                                            |    20(+0) |    0(+0) |   0(+0) |
| mbed-os/hal/mbed_gpio.o                                                                                                 |    88(+0) |    0(+0) |   0(+0) |
| mbed-os/hal/mbed_ticker_api.o                                                                                           |   950(+0) |    0(+0) |   0(+0) |
| mbed-os/hal/mbed_us_ticker_api.o                                                                                        |    64(+0) |    4(+0) |  65(+0) |
[... cut ...]
Total Flash memory (text + data): 34364(+0) bytes

Image: ./BUILD/NRF52_DK/GCC_ARM-RELEASE/LKAlphaOS-mbed.hex
```

This seems like a lot for an empty program... And I'm also surprised things like `lib_a-dtoa.o` are actually linked even if I'm only blinking a led or doing nothing at all.

As I said, I'm new to this, so maybe I'm doing something wrong, or maybe I missed the part in the documentation explaining why it's this way.

Is this the expected behavior?

Thanks a lot for your time! :)
Ladislas

40Grit, 2019-08-26:
Welcome aboard.

Without digging to deep into the particular items you brought up; There is currently a lot of refactoring in progress to address issues like this.

You might also want to try the bare metal "build profile"

It should eliminate a lot of the extra stuff and give you more insight.

ladislas, 2019-08-26:
Thanks @40Grit for the quick reply :)

> Without digging to deep into the particular items you brought up; There is currently a lot of refactoring in progress to address issues like this.

That's good news! Do you need help for that? 

> You might also want to try the bare metal "build profile"

What do you mean by that exactly? 

I'm using the following in my `mbed_app.json`

```
{
	"requires": ["bare-metal", "nordic"]
}
```

Do I need to add something more?

40Grit, 2019-08-26:
I'm not with ARM but since this is an open source project, we can use all the help we can get!

Looks like you already found bare-metal or your example is using it by default.

@0xc0170  any insight on code size reduction efforts?

0xc0170, 2019-08-26:
cc @ARMmbed/mbed-os-core

evedon, 2019-09-04:
A lot of work is being done in order to reduce the memory footprint.
For example, Mbed OS 5.14 will provide a minimal printf library to replace the standard printf calls with a smaller implementation. This typically saves several tens of KB of Flash.

ladislas, 2019-09-05:
Thanks a lot @evedon for the information, this is great to hear!

LDong-Arm, 2020-06-01:
Hi @ladislas,

In addition to bare-metal (no RTOS support) and minimal-printf as mentioned above, we also have _small C lib_ (for non-RTOS only) and _minimal-console_ (if you just need a console to print/get inputs without file handling) to reduce code size.

With all of them enabled, the `mbed_app.json` looks like
```json
{
    "requires": ["bare-metal"],
    "target_overrides": {
      "*": {
            "platform.stdio-minimal-console-only": true,
            "target.c_lib": "small"
        }
    }
}
```
(change `c_lib` to `default_lib` if you use Mbed OS 5.15 or earlier).
This reduces the static RAM usage to ~1KB and the flash usage to ~15KB for an empty application on NRF52_DK and GCC.

Further optimisations (e.g. getting fewer objects linked) are an ongoing effort.

evedon, 2020-06-01:
Considering this as resolved. As mentioned, further optimisations are an ongoing effort.

closed 2020-06-01 by evedon