ARMmbed/mbed-cli #916: mbed cli v1.10.0 - gcc_arm configs needs to have /bin in path, others not

JanneKiiskila, 2019-08-27:
Repro steps:

mbed --version
1.10.0

Running Ubuntu 16.04.

Download and get `gcc arm` : https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

Unarchive it for example under your user folder, i..e `/home/user`, for example

```
jankii01@ubuntu:~/gcc-arm-none-eabi-8-2019-q3-update$ pwd
/home/jankii01/gcc-arm-none-eabi-8-2019-q3-update
```

go to one of your mbed examples and configure mbed cli's path to gcc arm;

```
jankii01@ubuntu:~/mbed/mbed-client-testapp$ mbed config -G GCC_ARM_PATH "/home/jankii01/gcc-arm-none-eabi-8-2019-q3-update"
```

Try compiling, it will fail.

```
jankii01@ubuntu:~/mbed/mbed-cloud-client-example$ mbed config -G GCC_ARM_PATH "/home/jankii01/gcc-arm-none-eabi-8-2019-q3-update"
[mbed] /home/jankii01/gcc-arm-none-eabi-8-2019-q3-update now set as global GCC_ARM_PATH
jankii01@ubuntu:~/mbed/mbed-cloud-client-example$ mbed compile -m K64F -t GCC_ARM
[mbed] Working path "/home/jankii01/mbed/mbed-cloud-client-example" (program)
[mbed] WARNING: Missing Python modules were not auto-installed.
       The Mbed OS tools in this program require the following Python modules: hidapi, pywin32, wmi
       You can install all missing modules by running "pip install -r requirements.txt" in "/home/jankii01/mbed/mbed-cloud-client-example/mbed-os"
       On Posix systems (Linux, etc) you might have to switch to superuser account or use "sudo"
---
Could not find executable for GCC_ARM.
Currently set search path: /home/jankii01/gcc-arm-none-eabi-8-2019-q3-update
[mbed] ERROR: "/usr/bin/python" returned error.
       Code: 2
       Path: "/home/jankii01/mbed/mbed-cloud-client-example"
       Command: "/usr/bin/python -u /home/jankii01/mbed/mbed-cloud-client-example/mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build ./BUILD/K64F/GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

Add the `/bin` to the path, and it suddenly works.

```
jankii01@ubuntu:~/mbed/mbed-cloud-client-example$ mbed config -G GCC_ARM_PATH "/home/jankii01/gcc-arm-none-eabi-8-2019-q3-update/bin"
[mbed] /home/jankii01/gcc-arm-none-eabi-8-2019-q3-update/bin now set as global GCC_ARM_PATH
jankii01@ubuntu:~/mbed/mbed-cloud-client-example$ mbed compile -m K64F -t GCC_ARM
[mbed] Working path "/home/jankii01/mbed/mbed-cloud-client-example" (program)
[mbed] WARNING: Missing Python modules were not auto-installed.
       The Mbed OS tools in this program require the following Python modules: hidapi, pywin32, wmi
       You can install all missing modules by running "pip install -r requirements.txt" in "/home/jankii01/mbed/mbed-cloud-client-example/mbed-os"
       On Posix systems (Linux, etc) you might have to switch to superuser account or use "sudo"
---
[Warning] @,: Compiler version mismatch: Have 8.3.1; expected version >= 6.0.0 and < 7.0.0
Building project mbed-cloud-client-example (K64F, GCC_ARM)
Scan: mbed-cloud-client-example
Using ROM regions bootloader, header, application in this build.
  Region bootloader: size 0x8000, offset 0x0
  Region header: size 0x70, offset 0xa000
  Region application: size 0xf5c00,
```

However, IAR and ARM C work differently - there you may not add `/bin` to the path, or they fail. 

**Requirement: all compiler paths should work in the same way.**

JanneKiiskila, 2019-08-27:
This type of config works for me;

```
jankii01@ubuntu:~/mbed/mbed-cloud-client-example$ mbed config --list
[mbed] Global config:
ARM_PATH=/usr/local/ARM_Compiler_5.06u6/
GCC_ARM_PATH=/home/jankii01/gcc-arm-none-eabi-8-2019-q3-update/bin
IAR_PATH=/home/jankii01/IAR/IAR-BuildLx-Arm-8.32.1
ARMC6_PATH=/home/jankii01/ARMCompiler6.12
```

JanneKiiskila, 2019-08-27:
Seems I'm mistaken, ARM C 6 also seems to insist on having that `/bin` in the path.

SeppoTakalo, 2019-08-28:
No configuration is needed if you set Arm GCC into your path

JanneKiiskila, 2019-08-28:
I would not prefer to have them in the path, if you need to use multiple versions of the compilers - it's easier just to use `mbed config` rather than modify the path.

SeppoTakalo, 2019-11-08:
Btw, all these are in documentation:
https://os.mbed.com/docs/mbed-os/v5.14/tools/manual-installation.html

If you check the example table, GCC and Armc6 both have `/bin` at the end.

mark-edgeworth, 2019-12-04:
As this is covered by the documentation I'm closing.....

closed 2019-12-04 by mark-edgeworth