ARMmbed/mbed-cli #917: Generated includes file incorrectly prioritizes user headers over system version

petewarden, 2019-08-29:
The TensorFlow Lite source tree contains a header file with the name "string.h", inside the folder tensorflow/lite/. When compiling an mbed project with this nested file, any code that does `#include <string.h>` will import this version instead of the expected system header, causing mysterious compiler errors early in the process.

Other users have encountered the same problem: https://os.mbed.com/questions/85362/include-path-uses-incorrect-search

I thought I'd found a workaround by using .mbedignore to exclude the tensorflow folder and subdirectories, but while that gets around the earlier compilation errors, it has the unfortunate side effect of also not compiling the code in those folders.

After some digging, I found that the include paths were being set by a file like `@./BUILD/DISCO_F746NG/GCC_ARM/.includes_18349b8ce7da3e19df69aba86e441bb6.txt`. This contains a long list of `-I` directives, apparently for every subfolder in the workspace. There are a few problems with this:

 - Using `-I` means that every file is treated as a candidate for angle-bracket (system) includes. This is not what developers would normally expect for their own headers.
 - Recursively searching every subfolder is pretty surprising behavior too.
 - All of these directives override the default system headers, since they're searched first.

I would love any suggestions on workarounds. Renaming the current offending string.h in our codebase might work in the short term, but it's likely that other headers will hit similar issues in the future.

0xc0170, 2019-09-04:
Thanks for reporting. This is known issue of our build system. All folders are being included and are  subject to errors like this. There should be at least one issue related to this one. Have you checked in cli/or in this repository?

> Using -I means that every file is treated as a candidate for angle-bracket (system) includes. This is not what developers would normally expect for their own headers.

What would you suggest ?

Besides valid points above, TensorFlow should fix the name clash with the standard library, I would not expect `string.h` in their tree. Or if there already is, I would use relative inclusion to avoid surprises like this, `"lite/string.h"` in the tree. This does not fix the issue with our build system though as noted above (headers name clash in the tree).

petewarden, 2019-09-04:
> Or if there already is, I would use relative inclusion to avoid surprises like this, "lite/string.h" in the tree. 

This is already the case, we always include it via `#include "tensorflow/lite/string.h"`. The problem is that other system headers (for example cstring.h) `#include <string.h>`, and the current setup means the TensorFlow version is chosen instead of the system version.

Can we just use `-iquote` instead of `-I` for user-supplied code? https://gcc.gnu.org/onlinedocs/gcc-9.2.0/gcc/Directory-Options.html#Directory-Options

That will avoid the problem of user headers being in the system header search path.

adbridge, 2019-11-01:
Apologies, this was closed in error by the automated bot - re-opening.

djaenicke, 2020-01-14:
Is there any update on this issue?

dardeshna, 2022-10-19:
Is there a fix for this issue? I am having a similar problem with a header file in a third party library ([ros_lib_kinetic](https://os.mbed.com/users/garyservin/code/ros_lib_kinetic/)) named `time.h` which is messing up the `#include <time.h>` directives within the mbed os source code.

Only the root folder of the ros_lib_kinetic library needs to be passed to the compiler via `-I`. All the internal `#include` directives within the library are namespaced properly with relative paths.  Is there a way to disable this recursive behavior and manually set the directories passed with `-I`?

0xc0170, 2022-10-19:
@dardeshna if you use mbed tools (based on CMake), then this problem should not be there.

However, with mbed cli, the problem persists and it won't be fixed as the tools were frozen long time ago (deprecated).