ARMmbed/mbed-cli #920: No relative path/No quotes around path with spaces

mdeweerd, 2019-09-13:
The Makefile that is generated does not add quotes around paths and/or does use relative paths.

After executing on my Win10 machine, the following command for generating the makefile:
````
   mbed export -vv -i GCC_ARM -m NUCLEO_F091RC --profile debug
````

Where `mbed --version` returns `1.8.3`

The makefile shows (project name and username replaced with <...>):

````
INCLUDE_PATHS += -I../c:/Users/<user>/Mbed Programs/<project>/mbed-os
ASM_FLAGS += -c:/Users/<user>/Mbed Programs/<project>/mbed-os
````

This filed compilation, I had to add the quotes:
````
INCLUDE_PATHS += -I../"c:/Users/<user>/Mbed Programs/<project>/mbed-os"
ASM_FLAGS += -I"c:/Users/<user>/Mbed Programs/<project>/mbed-os"
````

Notice however that the first path is a mix of an absolute and a relative path.

Most pats are relative like this:

````
INCLUDE_PATHS += -I.././mbed-os/usb/device/utilities/events
````

So it seems that the makefile should have read the following instead:

````
INCLUDE_PATHS += -I.././mbed-os
ASM_FLAGS += -I"././mbed-os"
````
Or similar, or determine the absolute path from the Makefile path.

mdeweerd, 2019-10-31:
IMHO these are defects.

Reasons:

1. The default path including the space "Mbed programs" is proposed by the tool itself;
2. As a result the generated Makefile is invalid for certain steps.
3. Inconsistency (backward slash instead of a forward slash) breaking the makefile is a defect.
4. The suggestions were only there to indicate how this can be solved.

I am not using MBED anymore at this time - I am not reporting this issue more than I am now.

adbridge, 2019-11-01:
Apologies, this was closed in error by the automated bot - re-opening.

mdeweerd, 2022-01-06:
Closing this as it has been stale for over 2 years to clean up my issues list.

closed 2022-01-06 by mdeweerd