ARMmbed/mbed-cli #921: "notify" parameter missing in "project.py"

mdeweerd, 2019-09-17:
I had to add notify to ".tem/tools/project.py"

```
        try:
            export(
                mcu,
                ide,
                build=options.build,
                src=options.source_dir,
                macros=options.macros,
                project_id=options.program,
                zip_proj=not bool(options.source_dir) or options.zip,
                build_profile=profile,
                app_config=options.app_config,
                export_path=options.build_dir,
                notify=TerminalNotifier(),
                ignore=options.ignore
            )
```

Without that I get the following error when exporting (personal project path replaced with "."):

```
Traceback (most recent call last):
  File ".\.temp\tools\project.py", line 424, in <module>
    main()
  File ".\.temp\tools\project.py", line 416, in main
    ignore=options.ignore
TypeError: export() got an unexpected keyword argument 'xnotify'
[mbed] Working path "." (library)
[mbed] Program path "."
[mbed] Exec "C:\Python27\python.exe -u .\.temp\tools\project.py -i gcc_arm -m NUCLEO_F030R8 --source . --profile debug" in "."
[mbed] ERROR: "C:\Python27\python.exe" returned error.
       Code: 1
       Path: "."
       Command: "C:\Python27\python.exe -u .\.temp\tools\project.py -i gcc_arm -m NUCLEO_F030R8 --source . --profile debug"
       Tip: You could retry the last command with "-v" flag for verbose output
```

Version:
```
mbed --version
1.8.3
```

mark-edgeworth, 2019-09-18:
It looks like this issue may have already been resolved in later versions of the mbed-cli tooling, but is not in place for the legacy tooling used by mbed-2. Thanks for the report.

closed 2019-10-31 by ciarmcom

mdeweerd, 2019-10-31:
Closing this issue demonstrates a major issue with MBED: backward compatibility support.
MBED5 is not compatible with the platform we used, this is why we turned to MBED2.
This is one of the reasons that we dropped MBED.

0xc0170, 2019-11-04:
Hello @mdeweerd , we should strive for backward compatibility as long as we can. If there is a breakage, it should be reported (captured in the release notes and easily identifiable). However, there are cases like this one (I would like to know more how this one got in).

> MBED5 is not compatible with the platform we used, this is why we turned to MBED2.

Do you have more specific examples what was not compatible?

This can be reopened as valid bug. More information why it was closed: https://os.mbed.com/blog/entry/Improving-the-management-of-community-in/

0xc0170, 2019-11-04:
I think I found it.

This commit touched the main function and removed notify from export https://github.com/ARMmbed/mbed-os/commit/44030c23d37a15f787a7cc22a820f63223f4b4f4. I can't find a reason for that removal but might be just "a miss" there. as I see notify used in other export functions.

How can we reproduce locally this one ?