ARMmbed/mbed-cli #922: Export of sample program to CMake will not compile

robotdad, 2019-09-19:
Windows 10 OS, latest mbed cli release.

mbed import https://github.com/ARMmbed/mbed-os-example-blinky
cd mbed-os-example-blinky
mbed export --target NUCLEO_F411RE --ide cmake_gcc_arm
mkdir out
cd out
cmake .. -G Ninja -DCMAKE_TOOLCHAIN_FILE=C:\source\repos\arm-cmake-toolchains\arm-gcc-toolchain.cmake -DCMAKE_BUILD_TYPE=Debug

The toolchain file just sets the ARM GCC location. Compile fails with the three most recent ARM GCC toolchain releases
7-2017-q4-major
arm-none-eabi-g++.exe: error: unrecognized command line option '-rdynamic'

8-2018-q4-major (this looks like an ARM compiler bug, there are related issues filed)
arm-none-eabi-objcopy: mbed-os-example-blinky.hex 64-bit address 0x4b4fa308000000 out of range for Intel Hex file

8-2019-q3-update
arm-none-eabi-g++.exe: error: unrecognized command line option '-rdynamic'

I can find no way to control where the -rdymaic flag is being inserted here. I don't see anything in the CMakeLists.txt setting that.

The other thing is why is a blink sample pulling in the world to compile? This takes over 10 mins to compile before hitting the error in linking.

madchutney, 2019-09-23:
Hi @robotdad, just to confirm that I understand the issue. The CMake export is completing and the Ninja generation works as expected, but the result is not compatible with the most recent Arm GCC releases?

robotdad, 2019-09-23:
That is correct, it is also not compatible with the previous two releases of the Arm GCC releases as described in the initial report.

madchutney, 2019-09-24:
Thanks for raising the issue, we currently only officially suport 6-2017-q1 (https://os.mbed.com/docs/mbed-os/v5.13/tools/index.html) but the plan is to support 8-2019-q3 in 5.14.

robotdad, 2019-09-24:
I see the same build failure with 6-2017-q1 "unrecognized command line option '-rdynamic'"

0xc0170, 2019-09-25:
@robotdad Where is `rdynamic` coming from? I don't see this option set in our profiles (`tools/profiles`) ?

robotdad, 2019-09-25:
Its coming during the build when running ninja. I'm attaching the CMake file that was generated by the mbed cli and the Ninja build files generated by the above command.
[CMakeLists.txt](https://github.com/ARMmbed/mbed-cli/files/3654749/CMakeLists.txt)
[build.ninja.txt](https://github.com/ARMmbed/mbed-cli/files/3654757/build.ninja.txt)
[mbed-os-example-blinky_pp.link_script.ld.txt](https://github.com/ARMmbed/mbed-cli/files/3654845/mbed-os-example-blinky_pp.link_script.ld.txt)

Also, my toolchain file.
[arm-gcc-toolchain.cmake.txt](https://github.com/ARMmbed/mbed-cli/files/3654936/arm-gcc-toolchain.cmake.txt)

This is the output where the error occurs. I cannot figure out what is causing rdynamic to get passed. All I did was create the blink app as described above and try to build it. I have not modified any of the generated files.

[870/870] Linking CXX executable mbed-os-example-blinky                                                                 FAILED: mbed-os-example-blinky mbed-os-example-blinky_pp.link_script.ld
cmd.exe /C "cmd.exe /C "cd /D C:\source\mbed\mbed-os-example-blinky && arm-none-eabi-cpp -E -P -Wl,--no-export-dynamic -Wl,--gc-sections -Wl,--wrap,main -Wl,--wrap,_malloc_r -Wl,--wrap,_free_r -Wl,--wrap,_realloc_r -Wl,--wrap,_memalign_r -Wl,--wrap,_calloc_r -Wl,--wrap,exit -Wl,--wrap,atexit -Wl,-n -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x80000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0x20000 -DMBED_BOOT_STACK_SIZE=1024 ./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F411xE/device/TOOLCHAIN_GCC_ARM/STM32F411XE.ld -o C:/source/mbed/mbed-os-example-blinky/out/mbed-os-example-blinky_pp.link_script.ld && cd C:\source\mbed\mbed-os-example-blinky\out" && C:\armgcc\6-2017-q1-update\bin\arm-none-eabi-g++.exe --sysroot=C:/armgcc/6-2017-q1-update/bin/../arm-none-eabi  -std=gnu++14 -fno-rtti -Wvla -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -O0 -g3 -DMBED_DEBUG -DMBED_TRAP_ERRORS_ENABLED=1 -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x80000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0x20000  -include mbed_config.h -g -Wl,--gc-sections -Wl,--wrap,main -Wl,--wrap,__malloc_r -Wl,--wrap,__free_r -Wl,--wrap,__realloc_r -Wl,--wrap,__memalign_r -Wl,--wrap,__calloc_r -Wl,--wrap,exit -Wl,--wrap,atexit -Wl,-n -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x80000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0x20000 -DMBED_BOOT_STACK_SIZE=1024  -Wl,--start-group -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys  -Wl,--end-group -T C:/source/mbed/mbed-os-example-blinky/out/mbed-os-example-blinky_pp.link_script.ld -Wl,--export-dynamic -rdynamic @CMakeFiles\mbed-os-example-blinky.rsp  -o mbed-os-example-blinky  && cmd.exe /C "cd /D C:\source\mbed\mbed-os-example-blinky\out && arm-none-eabi-objcopy -O ihex C:/source/mbed/mbed-os-example-blinky/out/mbed-os-example-blinky C:/source/mbed/mbed-os-example-blinky/out/mbed-os-example-blinky.hex && "C:\Program Files\CMake\bin\cmake.exe" -E echo "-- built: C:/source/mbed/mbed-os-example-blinky/out/mbed-os-example-blinky.hex"""
arm-none-eabi-g++.exe: error: unrecognized command line option '-rdynamic'
ninja: build stopped: subcommand failed.

0xc0170, 2019-09-26:
What is the version of your cmake? I believe this is an error in cmake rather than tools here.

`--export-dynamic` seems to be valid, but `rdynamic` not. The only related thing in our cmake template is: `SET_TARGET_PROPERTIES({{name}} PROPERTIES ENABLE_EXPORTS 1)`

See reports like https://cmake.org/Bug/print_bug_page.php?bug_id=15630, might help

0xc0170, 2019-09-26:
see https://stackoverflow.com/questions/10599038/can-i-skip-cmake-compiler-tests-or-avoid-error-unrecognized-option-rdynamic

robotdad, 2019-09-26:
cmake version 3.12.4

Per the second link I added before the project call in CMakeLists.txt
set(CMAKE_SHARED_LIBRARY_LINK_C_FLAGS "")
set(CMAKE_SHARED_LIBRARY_LINK_CXX_FLAGS "")

No change, rdynamic is still being passed.

As it was mentioned in that second post the compiler checks when running CMake are passing.

robotdad, 2019-10-31:
This is not a discussion item. This is a defect. I am using the compiler that is said to be supported. I have tried the suggestions mentioned here and they do not work.

adbridge, 2019-11-01:
This was closed by the automated bot by mistake - re-opening

gabewillen, 2019-12-02:
this issue is resolved for creating an executable by commenting out 
# SET_TARGET_PROPERTIES(${PROJECT_NAME} PROPERTIES ENABLE_EXPORTS 1)

0xc0170, 2020-03-17:
I dont understand why there is `ENABLE_EXPORTS` as exported project is the final application