ARMmbed/mbed-cli #927: using cli not downloading mbed

D252874, 2019-10-21:
i am just started using the cli for testing and development purpose i am waiting for it to download for longer than 20 min already then i thought i do some testing of your netwerk and when i ping mbed.com or os.mbed.com i get an timeout is that something with your protection or is it something with me with i doubt becouse any other website i ping gets an response in 4 a 5 ms

D252874, 2019-10-21:
[image]
[image]
here are the testing i did and my cli

loverdeg-ep, 2019-10-21:
This seems to be a common issue.

My teams solution has been to keep a mirror of mbed on our local network and clone it "manually" outside of mbed-cli.

D252874, 2019-10-21:
that sound great but does that also work when you have't gotten it yet like download it from there website and then use that or doesn't that work ?

loverdeg-ep, 2019-10-21:
Not sure. 
We don't use mbed-cli directly except for build/compile.

D252874, 2019-10-22:
so now it is an day later and I still can't ping mbed maby something with security or some weird thing but now it says =
[image]
and I tryed running it as an admin and tryed it without windows explorer

40Grit, 2019-10-22:
@mark-edgeworth @0xc0170

mark-edgeworth, 2019-10-22:
os.mbed.com doesn't seem to respond to ping, but it is alive (check it in a browser). Sounds like the earlier attempt to download mbed failed or was killed, and has now left behind an open file. Suggest that you check for open filehandles, or restart the machine might help...

D252874, 2019-10-22:
could it be possible that it thinks it has it in the cache but it is not in the cache so it crashed

D252874, 2019-10-22:
tested it and now i get this =
[image]

D252874, 2019-10-23:
this looks a lot like issue [#436](https://github.com/ARMmbed/mbed-cli/issues/436) but i don't know what the problem is. and how to fix it
i did mbed deploy in an folder i created got this =
[image]
with an new problem with tagg --mbedlib

D252874, 2019-10-23:
no i am making an project without --mbedlib and i get kind of an progress bar that i am not getting with the mbedlib i did do this with the very embose but it seems to work i am not getting no error. 

but found out i can't use this for the export i want for some reason
i will keep this here becouse i am maby gone use mbed cli but i need this to work for that

closed 2019-10-31 by ciarmcom