ARMmbed/mbed-cli #930: mbed-cli is broken on MacOS Catalina

iotexpert, 2019-11-03:
When I installed MacOS Catalina the MBEDCLI stopped working.

I made a HACK by changing the path to the Terminal.app in the file:

The file /Applications/MBEDCLI.app/Contents/Resources/script

# open Terminal and call run-mbed-cli.sh, use Popen instead of subprocess.call to pass in environment variables
subprocess.Popen(
    ["/usr/bin/open", "--fresh", "-a", "/System/Applications/Utilities/Terminal.app", "source/run-mbed-cli.sh"]
    )

SeppoTakalo, 2019-11-08:
I don't understand.. What exactly stopped working.

I'm using Mac OS X Catalina, and I have not modified mbed-cli. Works like it should. I'm using both Python2 and Python3. Both seems to work.

iotexpert, 2019-11-21:
Take a mac on an old os
install mbed-cli (the installer)
Update to macos catalina
run the "Mbed CLI" app
See it fail

40Grit, 2019-11-21:
Just one of the many reasons I wish there was an official docker container for Mbed Tooling.

maclobdell, 2019-12-13:
@SeppoTakalo - did you install mbed cli with the Mac installer?  Could this be an installer issue?

arshadpc, 2019-12-26:
is this resolved if yes kindly let me know the solution?

arshadpc, 2020-01-03:
@SeppoTakalo kindly answer above query?

0xc0170, 2020-01-03:
@arshadpc Are you having the same issue?  Did you use installer? If you install it via pip, does it fail the same way?

arshadpc, 2020-01-06:
> @arshadpc Are you having the same issue? Did you use installer? If you install it via pip, does it fail the same way?
@0xc0170  Yes ,i have not tried with pip i used installer and seeing issue.

SeppoTakalo, 2020-01-07:
No, I have not used the installer.

For mbed-cli and related tools, I have always installed those with clean Python virtual environment.

SeppoTakalo, 2020-01-07:
So is the conclusion then that installer is broken, and with `pip install` it works.

emha69, 2020-01-10:
> When I installed MacOS Catalina the MBEDCLI stopped working.

The path to Terminal.app is not correct, the error displayed is:

​Printing from run-mbed-cli.py
 MBED_CLI_APP_DIR=/Applications/MBEDCLI.app/Contents/Resources
$HOME= /Users/<user>
FSPathMakeRef(/Applications/Utilities/Terminal.app) failed with error -43.

> I made a HACK by changing the path to the Terminal.app in the file:
>
> The file /Applications/MBEDCLI.app/Contents/Resources/script
> # open Terminal and call run-mbed-cli.sh, use Popen instead of [...]

After adding "/System" folder in front of "/Applications" as described the script works as expected.

iotexpert, 2020-01-10:
When you run the mac installer it doesn’t work correctly… you have to fix it like I said

It works find when you setup  thing in a venv

Except you might want to put in instruction about how to get the mac to run the compiler things with the security setting problem

Alan

emha69, 2020-01-11:
I can confirm that fix as suggested by Alan works. I think this is more a PATH issue as an issue of installer. Looks like AppIe has relocated some system apps with recent OS upgrade. I am running 10.15.2.

I have installed everything with mac installer without any problems, then applied fix as suggested, but had to install mbed-cli afterwards with pip manually.

To summarize, the mac installer works, but the installation is not complete and requires some additional
manual steps.

iotexpert, 2020-01-12:
> … <x-msg://5/#>

Correct

Alexcornila, 2020-01-13:
I am a noob at this had the CLI tools and arm compiler working on my Mac before, I have tried the OP suggestion but doesn’t work for me. Any other suggestion to get this working?

emha69, 2020-01-13:
> I am a noob at this had the CLI tools and arm compiler working on my Mac before, I have tried the OP suggestion but doesn’t work for me. Any other suggestion to get this working?

I had Python3.8 and GNU ARM v9.2.1 already installed on the system, before running mbed-cli installer, so I don't know what happens, if these two are not installed. Maybe you should try with installing the Python3.8 and add it to the PATH.

Alternatively, try to uninstall old mbed-cli, then upgrade the macos and install mbed-cli again.

thesulaj, 2020-07-18:
I found a (hacky) solution to this. The problem is that the installer is unable to find the terminal app, as it has moved from `/Applications/Utilities/Terminal.app` to `/System/Applications/Utilities/Terminal.app`. So, we simply need to update the path.

1. Navigate to the resources directory in MBEDCLI.app: `cd /Applications/MBEDCLI.app/Contents/Resources`.
2. Replace `/Applications/Utilities/Terminal.app` with `/System/Applications/Utilities/Terminal.app` in the `./script` and `./run-mbed-cli.py` files. Both instances can be found on line 32 of the files.

After the above, the MBEDCLI.app app can be run as expected.