ARMmbed/mbed-cli #932: SSH Auth for git

loverdeg-ep, 2019-11-07:
I'm getting word from some other devs that SSH keys are not working with mbed .lib files

@aglass0fmilk

0xc0170, 2019-11-08:
It would be good to have a case to reproduce, even though this might be easy, just in case we follow the same steps.

+ your enviroment - what mbed cli version, git version, etc.

40Grit, 2019-11-08:
This was an end of the day issue.
Details to follow.

SeppoTakalo, 2019-11-08:
Sounds not plausible.
Mbed call `git` normally. So if your git setup is correct, it should work.

Now, the problem might be that it does not use SSH by default. I always clone things like this: `mbed deploy --protocol ssh`

Sometimes Mbed-cli tries to use HTTP first.

40Grit, 2019-11-11:
Why would it not just infer from the url that it should use ssh?

@aglass0fmilk

loverdeg-ep, 2019-11-11:
@trowbridgec Any further info on this?

trowbridgec, 2019-11-11:
I was able to get this to work by changing the path from something like this
```
ssh://gitlab.local/path.to.repo.git/#0123456789
```
to something like this
```
git@gitlab.local:path.to.repo.git/#0123456789
```
for our internal GitLab instances.