ARMmbed/mbed-cli #935: Application build omits custom target includes and object files

bikeNomad, 2019-11-26:
I want to build several mbed programs using the same processor, the STM32F072VB.

This processor is a 100-pin version of the 64-pin STM32F072RB processor that is
in the existing `NUCLEO_F072RB` target.

I only needed to add some definitions for missing pins and change some pin names.

However, my attempt at using `--custom-targets` has not worked as I expected.

I had both problems with include directory ordering and linker response scripts.

I'm using `mbed-cli` version 1.10.1 .

## Include directory ordering problem

My new custom target directories were not being searched before the `mbed-os`
directory of the target that it inherits from.

The existing `mbed-os/targets/TARGET_STM` hierarchy looks like this (just showing
directories and the files I wanted to change):

    TARGET_STM
    └── TARGET_STM32F0
        ├── TARGET_NUCLEO_F072RB
        │   ├── PeripheralNames.h
        │   ├── PeripheralPins.c
        │   ├── PinNames.h
        │   ├── objects.h
        │   └── device
        │       ├── TOOLCHAIN_ARM_MICRO
        │       ├── TOOLCHAIN_ARM_STD
        │       ├── TOOLCHAIN_GCC_ARM
        │       └── TOOLCHAIN_IAR
        └── device

My custom targets directory (called `Targets`) looks like this:

    Targets
    ├── custom_targets.json
    ├── mbed_lib.json
    ├── mbed_settings.py
    └── targets
        └── TARGET_STM
            └── TARGET_STM32F0
                └── TARGET_STM32F072VB
                    ├── PeripheralNames.h
                    ├── PeripheralPins.c
                    ├── objects.h
                    └── PinNames.h

`Targets/custom_targets.json` looks like this:

```json
{
  "STM32F072VB": {
    "inherits": ["NUCLEO_F072RB"],
    "extra_labels_add": ["STM32F072VB"],
    "device_name": "STM32F072VB",
    "default_lib": "small",
    "supported_form_factors": [],
    "features": [],
    "components": [],
    "overrides": {
        "clock_source": "USE_PLL_HSI"
    }
  }
}
```

My working directory is organized like this:

    ├── Common      common code used by MyProgram and others at this level, also my custom build profile
    ├── MyProgram       where main.cpp lives
    ├── Targets     my custom-targets directory, described above
    ├── common-lib-build    where the code in Common gets built into libcommon.a
    ├── myprogram-build where MyProgram gets built
    ├── mbed-os     stock mbed library (from mbed import mbed-os)
    └── mbed-os-build   where mbed-os gets built into libmbed-os.a

When I looked at the `mbed-os-build/.includes_xxxxx.txt` file and the verbose
compilation command lines, I saw that my target include directories weren't
included at all. The end of the `.includes` file looked like this:

    "-Imbed-os/targets/TARGET_STM"
    "-Imbed-os/targets/TARGET_STM/TARGET_STM32F0"
    "-Imbed-os/targets/TARGET_STM/TARGET_STM32F0/TARGET_NUCLEO_F072RB"
    "-Imbed-os/targets/TARGET_STM/TARGET_STM32F0/TARGET_NUCLEO_F072RB/device"
    "-Imbed-os/targets/TARGET_STM/TARGET_STM32F0/device"

So I added the two important include directories to
my `mydebug.json` build profile (though I don't think this should be necessary):

```json
{
    "GCC_ARM": {
        "common": [ "-ICommon", "-ITargets/targets/TARGET_STM/TARGET_STM32F0/TARGET_STM32F072VB" ],
        "c": ["-std=gnu11"],
        "cxx": ["-std=gnu++14", "-fno-rtti", "-fno-exceptions" ]
    }
}
```

Now the compiler command lines have my include directories preceding the `mbed-os` directories.

I built `mbed-os-build/libmbed-os.a` (without any reported errors) like this:

```bash
MBED_OS_DIR=mbed-os mbed compile -t GCC_ARM -m STM32F072VB --build mbed-os-build \
    --library --source mbed-os --custom-targets Targets \
    --profile debug --profile Common/mydebug.json
```
And then I built my app like this:

```bash
MBED_OS_DIR=mbed-os mbed compile -t GCC_ARM -m STM32F072VB --build myprogram-build \
    --source MyProgram --custom-targets Targets \
    --source common-lib-build --source mbed-os-build \
    --profile debug --profile Common/mydebug.json
```

However, the pin maps were still not correct, because my `PeripheralPins.c` file
was not being used in the resultant app.

## Library linking issues

Adding `-Wp,-H` to the gcc command line (in `Common/mydebug.json`) to trace
includes and adding `-v` to the `mbed compile` command showed me that the
`mbed-os` `NUCLEO_F072RB/PeripheralPins.c` file was being compiled into the resultant
library instead of my custom target version, in spite of my providing the
source file in my custom target directory.

So I added `--source Targets` to my app's compile line:

```bash
MBED_OS_DIR=mbed-os mbed compile -t GCC_ARM -m STM32F072VB --build myprogram-build \
    --source MyProgram --source Targets --custom-targets Targets \
    --source common-lib-build --source mbed-os-build \
    --profile debug --profile Common/mydebug.json
```

This at least got my custom target's `PeripheralPins.c` file compiled along with
the app's `main.cpp`.

However, the link ordered all the `mbed-os` object files before mine, despite
also linking against `libmbed-os.a`:

    ... many mbed-os object files ...
    mbed-os-build/mbed-os/targets/TARGET_STM/TARGET_STM32F0/TARGET_NUCLEO_F072RB/PeripheralPins.o
    ... many more mbed-os object files ...
    myprogram-build/MyProgram/main.o
    myprogram-build/Targets/targets/TARGET_STM/TARGET_STM32F0/TARGET_STM32F072VB/PeripheralPins.o
    -Wl,--start-group
    -lCommon
    -lmbed-os
    -lstdc++
    -lsupc++
    -lm
    -lc
    -lgcc
    -lnosys
    -Wl,--end-group
    -T myprogram-build/.link_script.ld
    -L common-lib-build
    -L mbed-os-build
    -lCommon
    -lmbed-os
    -lstdc++
    -lsupc++
    -lm
    -lc
    -lgcc
    -lnosys

Editing the linker response file to remove all the `mbed-os-build/*.o` files
resulted in a correct link:

    -Wl,--gc-sections
    -Wl,--wrap,main
    -Wl,--wrap,_malloc_r
    -Wl,--wrap,_free_r
    -Wl,--wrap,_realloc_r
    -Wl,--wrap,_memalign_r
    -Wl,--wrap,_calloc_r
    -Wl,--wrap,exit
    -Wl,--wrap,atexit
    -Wl,-n
    --specs=nano.specs
    -mcpu=cortex-m0
    -mthumb
    -DMBED_BOOT_STACK_SIZE=1024
    -o myprogram-build/MyProgram.elf
    -Wl,-Map=myprogram-build/MyProgram.map
    myprogram-build/MyProgram/main.o
    myprogram-build/Targets/targets/TARGET_STM/TARGET_STM32F0/TARGET_STM32F072VB/PeripheralPins.o
    -Wl,--start-group
    -lCommon
    -lmbed-os
    -lstdc++
    -lsupc++
    -lm
    -lc
    -lgcc
    -lnosys
    -Wl,--end-group
    -T myprogram-build/.link_script.ld
    -L common-lib-build
    -L mbed-os-build
    -lCommon
    -lmbed-os
    -lstdc++
    -lsupc++
    -lm
    -lc
    -lgcc
    -lnosys

## What I'd like to know

How do I get my program to link against the previously compiled `libmbed-os.a`
without separately linking in all the mbed-os object files?

loverdeg-ep, 2019-11-26:
I believe you are interpreting '--custom-targets' correctly. It looks for the directory in which 'custom_targets.json'

I'm not sure you need to specify the '--source' directly though.

loverdeg-ep, 2019-11-26:
My team and I have been through this mess before. We contributed the --custom-targets option to try and make it a little easier.

bikeNomad, 2019-11-26:
My build `Makefile` currently deletes all the `*.o` files under `mbed-os-build`, which helps.

loverdeg-ep, 2019-11-26:
Oh are you exporting?

bikeNomad, 2019-11-26:
I tried to export to Eclipse, but `mbed-cli` told me that it couldn't find the linker script.

loverdeg-ep, 2019-11-27:
This won't help you much but,
In the past I was a slave to eclipse's indexer. I've grown fond of VS-Code. @janjongboom has a solid exporter with working intellisense etc. for VS-Code.

loverdeg-ep, 2019-12-05:
@bikeNomad  did you ever solve this?

bikeNomad, 2019-12-11:
@loverdeg-ep To make the library linking work, I added `Makefile` rules to remove the mbed `.o` files after building `libmbed-os.a`.

What I haven't been able to do is to get my customized linker script to be used. Since I've inherited my target from the `NUCLEO_F072RB`, its linker script is being used. I even tried to copy the `TARGET_NUCLEO_F072RB` hierarchy to my custom target, editing only the linker script, `Targets/targets/TARGET_STM/TARGET_STM32F0/TARGET_STM32F072VB/PeripheralPins.c` and `Targets/targets/TARGET_STM/TARGET_STM32F0/TARGET_STM32F072VB/PinNames.h`. But the link always uses the `mbed-os/targets/TARGET_STM/TARGET_STM32F0/TARGET_NUCLEO_F072RB/device/TOOLCHAIN_GCC_ARM/STM32F072XB.ld` linker script.

I have been trying to avoid making a branch of the `mbed-os-5.15` tree just to add my changes (we're using Subversion, and it's not easy to branch a git repository under `svn` version control without copying the whole tree.

closed 2019-12-11 by bikeNomad