ARMmbed/mbed-cli #944: setup.py install and mercurial version requirement

adamvm, 2019-12-23:
sudo python3.8 setup.py install

........

Copying mbed_cli-1.10.2-py3.8.egg to /usr/local/lib/python3.8/dist-packages
mbed-cli 1.10.2 is already the active version in easy-install.pth
Installing mbed script to /usr/local/bin
Installing mbed-cli script to /usr/local/bin

Installed /usr/local/lib/python3.8/dist-packages/mbed_cli-1.10.2-py3.8.egg
Processing dependencies for mbed-cli==1.10.2
Searching for mercurial>=5.2
Reading https://pypi.org/simple/mercurial/
Download error on https://pypi.org/simple/mercurial/: [Errno 13] Permission denied -- Some packages may not be found!
Couldn't find index page for 'mercurial' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading https://pypi.org/simple/
Download error on https://pypi.org/simple/: [Errno 13] Permission denied -- Some packages may not be found!
No local packages or working download links found for mercurial>=5.2
error: Could not find suitable distribution for Requirement.parse('mercurial>=5.2')

madchutney, 2020-01-07:
Could you please try installing `mbed-cli` using pip e.g. `pip install mbed-cli` and if this doesn't work try installing Mercurial individually e.g. `pip install Mercurial` and send us the results. Thanks

MarceloSalazar, 2020-01-15:
When trying to install mbed-cli on a fresh Windows PC with Python 3.7, I get an error related to the mercurial dependency
```
 pip install mbed-cli
Collecting mbed-cli
  Downloading https://files.pythonhosted.org/packages/05/9e/eca848d298700aba51d67b2fa4933106f23420436e6ad0e549ed9b78ccae/mbed_cli-1.10.2-py2.py3-none-any.whl (41kB)
     |████████████████████████████████| 51kB 825kB/s
Collecting mbed-os-tools<0.1.0,>=0.0.9 (from mbed-cli)
  Downloading https://files.pythonhosted.org/packages/89/4b/300579bd26825fde7293ff5c0114f1045b653bc84aaa670895764866c0e0/mbed_os_tools-0.0.12-py2.py3-none-any.whl (151kB)
     |████████████████████████████████| 153kB 1.6MB/s
Collecting pyserial<4.0,>=3.0 (from mbed-cli)
  Downloading https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl (193kB)
     |████████████████████████████████| 194kB 2.2MB/s
Collecting mercurial>=5.2 (from mbed-cli)
  Downloading https://files.pythonhosted.org/packages/f7/25/f0dba6943dfead28c07e0356f627cf4cd98842c6760593f1ea4115097ead/mercurial-5.2.2.tar.gz (7.3MB)
     |████████████████████████████████| 7.3MB 3.3MB/s
    ERROR: Command errored out with exit status 1:
     command: 'C:\Users\Marcelo\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Marcelo\\AppData\\Local\\Temp\\pip-install-w_mb8oeq\\mercurial\\setup.py'"'"'; __file__='"'"'C:\\Users\\Marcelo\\AppData\\Local\\Temp\\pip-install-w_mb8oeq\\mercurial\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base pip-egg-info
         cwd: C:\Users\Marcelo\AppData\Local\Temp\pip-install-w_mb8oeq\mercurial\
    Complete output (24 lines):
    Traceback (most recent call last):
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.1776.0_x64__qbz5n2kfra8p0\lib\site-packages\setuptools\msvc.py", line 489, in _find_latest_available_vc_ver
        return self.find_available_vc_vers()[-1]
    IndexError: list index out of range

    During handling of the above exception, another exception occurred:

    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\Marcelo\AppData\Local\Temp\pip-install-w_mb8oeq\mercurial\setup.py", line 1490, in <module>               name='mercurial.zstd', root=os.path.abspath(os.path.dirname(__file__))
      File "contrib/python-zstandard\setup_zstd.py", line 147, in get_c_extension
        compiler.initialize()
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.1776.0_x64__qbz5n2kfra8p0\lib\distutils\_msvccompiler.py", line 239, in initialize
        vc_env = _get_vc_env(plat_spec)
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.1776.0_x64__qbz5n2kfra8p0\lib\site-packages\setuptools\msvc.py", line 185, in msvc14_get_vc_env
        return EnvironmentInfo(plat_spec, vc_min_ver=14.0).return_env()
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.1776.0_x64__qbz5n2kfra8p0\lib\site-packages\setuptools\msvc.py", line 843, in __init__
        self.si = SystemInfo(self.ri, vc_ver)
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.1776.0_x64__qbz5n2kfra8p0\lib\site-packages\setuptools\msvc.py", line 485, in __init__
        self.vc_ver = vc_ver or self._find_latest_available_vc_ver()
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.1776.0_x64__qbz5n2kfra8p0\lib\site-packages\setuptools\msvc.py", line 492, in _find_latest_available_vc_ver
        raise distutils.errors.DistutilsPlatformError(err)
    distutils.errors.DistutilsPlatformError: Microsoft Visual C++ 14.0 is required. Get it with "Microsoft Visual C++ Build Tools": https://visualstudio.microsoft.com/downloads/
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

I understand there is no egg file in pypi hence it's asking to have MS Visual C++ installed. Same problem with Python 3.8 and 3.6

maclobdell, 2020-01-21:
@madchutney Any resolution to this issue?  It seems critical.  This is affecting many people.  I am also having this problem - see below.

```
C:\Users\user01>pip install -U mbed-cli --user
Collecting mbed-cli
  Using cached https://files.pythonhosted.org/packages/05/9e/eca848d298700aba51d67b2fa4933106f23420436e6ad0e549ed9b78ccae/mbed_cli-1.10.2-py2.py3-none-any.whl
Collecting mercurial>=5.2 (from mbed-cli)
  Using cached https://files.pythonhosted.org/packages/f7/25/f0dba6943dfead28c07e0356f627cf4cd98842c6760593f1ea4115097ead/mercurial-5.2.2.tar.gz
    ERROR: Command errored out with exit status 1:
     command: 'c:\users\user01\appdata\local\programs\python\python37\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\user01\\AppData\\Local\\Temp\\pip-install-1d_baic1\\mercurial\\setup.py'"'"'; __file__='"'"'C:\\Users\\user01\\AppData\\Local\\Temp\\pip-install-1d_baic1\\mercurial\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base pip-egg-info
         cwd: C:\Users\user01\AppData\Local\Temp\pip-install-1d_baic1\mercurial\
    Complete output (24 lines):
    Traceback (most recent call last):
      File "c:\users\user01\appdata\local\programs\python\python37\lib\site-packages\setuptools\msvc.py", line 489, in _find_latest_available_vc_ver
        return self.find_available_vc_vers()[-1]
    IndexError: list index out of range

    During handling of the above exception, another exception occurred:

    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\user01\AppData\Local\Temp\pip-install-1d_baic1\mercurial\setup.py", line 1490, in <module>
        name='mercurial.zstd', root=os.path.abspath(os.path.dirname(__file__))
      File "contrib/python-zstandard\setup_zstd.py", line 147, in get_c_extension
        compiler.initialize()
      File "c:\users\user01\appdata\local\programs\python\python37\lib\distutils\_msvccompiler.py", line 239, in initialize
        vc_env = _get_vc_env(plat_spec)
      File "c:\users\user01\appdata\local\programs\python\python37\lib\site-packages\setuptools\msvc.py", line 185, in msvc14_get_vc_env
        return EnvironmentInfo(plat_spec, vc_min_ver=14.0).return_env()
      File "c:\users\user01\appdata\local\programs\python\python37\lib\site-packages\setuptools\msvc.py", line 843, in __init__
        self.si = SystemInfo(self.ri, vc_ver)
      File "c:\users\user01\appdata\local\programs\python\python37\lib\site-packages\setuptools\msvc.py", line 485, in __init__
        self.vc_ver = vc_ver or self._find_latest_available_vc_ver()
      File "c:\users\user01\appdata\local\programs\python\python37\lib\site-packages\setuptools\msvc.py", line 492, in _find_latest_available_vc_ver
        raise distutils.errors.DistutilsPlatformError(err)
    distutils.errors.DistutilsPlatformError: Microsoft Visual C++ 14.0 is required. Get it with "Microsoft Visual C++ Build Tools": https://visualstudio.microsoft.com/downloads/
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
WARNING: You are using pip version 19.2.3, however version 20.0.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.

```

A workaround is to not install the latest version.

`pip install mbed-cli==1.10.1`

maclobdell, 2020-02-19:
@madchutney - please let us know if you have an update.

madchutney, 2020-02-20:
@maclobdell We believe this was caused by a problem with `pip` and it should now have been resolved. Can you please check that this is still occurring?

acabarbaye, 2020-02-20:
There was indeed an issue with `pip` a few weeks back #949 which made us think `pip` could be responsible for this isssue. Trying on my windows machine, `pip install mercurial` worked fine but actually, I just realised I have Microsoft Visual C++ installed. Trying on a separate Windows machine, I got the same error as you no what what version of `pip` I use. Looking a bit more closely at what is distributed on PyPI for [mercurial](https://pypi.org/project/mercurial/5.3/#files) I realised that there is no wheel for python 3 (only for cp27 hence python 2.7). Therefore, pip needs to build the package at installation, which means it needs a C++ compiler installed as there is some C/C++ code in mercurial. Until mercurial is packaged and released as a wheel for python3, it will not be possible to install it on Windows without requiring users of python3 to install a compiler on their machine.

MarceloSalazar, 2020-02-20:
We should really not require developers to install the C++ compiler as it's a massive package and impacts the UX to get started.
I hope we can have a wheel package for Python 3 soon and from there try to freeze the dependencies.

madchutney, 2020-02-20:
Yes hopefully, Python 3 support is still in beta for Windows according to this page https://www.mercurial-scm.org/wiki/Python3