ARMmbed/mbed-cli #945: mbed deploy auto installs missing Python modules

korjaa, 2020-01-03:
I thought the auto-installation was already removed. But maybe this got forgotten

https://github.com/ARMmbed/mbed-cli/blob/1.10.2/mbed/mbed.py#L1697

You can accidentally ruin your python environment with following commands:
```
$ pip install mbed-cli==1.10.2
...
$ cd mbed-os-projext-X
...
$ mbed deploy
[mbed] Working path "/home/jaakor01/workspace/mbed-cloud-client-example" (program)
[mbed] Adding library "mbed-cloud-client" from "ssh://git@github.com/ARMmbed/mbed-cloud-client" at rev #a90cdc723466
[mbed] Adding library "mbed-os" from "ssh://git@github.com/ARMmbed/mbed-os" at rev #64853b354fa1
[mbed] Adding library "drivers/COMPONENT_WIFI_ISM43362" from "ssh://git@github.com/ARMmbed/wifi-ism43362" at rev #1c040113be0f
[mbed] Auto-installing missing Python modules (pyyaml, jsonschema, mbed_cloud_sdk, jinja2, mbed_ls, mbed_host_tests, mbed_greentea, pyelftools, manifest_tool, icetea, pycryptodome, pyusb, cmsis_pack_manager, psutil, cryptography, click)...
```

korjaa, 2020-01-03:
Related to
https://github.com/ARMmbed/mbed-cli/issues/899
https://github.com/ARMmbed/mbed-cli/issues/892
https://github.com/ARMmbed/mbed-cli/issues/888

rmikhno, 2020-01-23:
@ciarmcom, 
Do you have any updates on this ? ...  thanks!

korjaa, 2020-01-23:
Hello @rmikhno,
No updates yet.

madchutney, 2020-02-20:
Hi @korjaa / @rmikhno,

We are aware that this is an ongoing problem with the architecture of the tooling. It is not a simple matter to change as python packages require can be different depending on the version Mbed OS, while the installed Mbed CLI has to work for all versions. We are adopting a different approach for new tooling which we hope will be ready for Mbed 6 which should eliminate this issue.