ARMmbed/mbed-cli #946: Documentation link is broken

edoardesd, 2020-01-08:
https://github.com/ARMmbed/mbed-cli/blob/master/README.md is not working

0xc0170, 2020-01-08:
Where is this one coming from (documentation url) ? I can see it uses .md extension, instead of .rst

mgrah, 2020-01-14:
Just open the .rst file and there's a link (see the original bug report) that doesn't work. 

https://github.com/ARMmbed/mbed-cli/blob/master/README.rst contains a link to https://github.com/ARMmbed/mbed-cli/blob/master/README.md 

The link is under the heading "Get Started" and the link text is "read the full documentation on GitHub". I'm sure that's not supposed to be a recursive link to itself. :)

mark-edgeworth, 2020-04-14:
fixed in https://github.com/ARMmbed/mbed-cli/pull/950

closed 2020-04-14 by mark-edgeworth