ARMmbed/mbed-cli #947: mbed device-management init does not seem to work

GuyWi, 2020-01-16:
I get this error:
```
C:\se\mbed-cloud-client-example-internal-poc-atmel-se>mbed device-management init -d arm.com --model-name k64f

[mbed] Working path "C:\se\mbed-cloud-client-example-internal-poc-atmel-se" (program)
Traceback (most recent call last):

  File "C:\se\mbed-cloud-client-example-internal-poc-atmel-se\mbed-os\tools\device_management.py", line 192, in <module>
    main()

  File "C:\se\mbed-cloud-client-example-internal-poc-atmel-se\mbed-os\tools\device_management.py", line 187, in main
    }[options.action](options) or 0

  File "C:\se\mbed-cloud-client-example-internal-poc-atmel-se\mbed-os\tools\device_management.py", line 120, in inner
    "key": api_key_value

  File "c:\python27\lib\site-packages\mbed_cloud\pagination.py", line 150, in next
    return self.__next__()

  File "c:\python27\lib\site-packages\mbed_cloud\pagination.py", line 159, in __next__
    raise StopIteration()
StopIteration

[mbed] ERROR: "c:\python27\python.exe" returned error.

       Code: 1

       Path: "C:\se\mbed-cloud-client-example-internal-poc-atmel-se"

       Command: "c:\python27\python.exe -u C:\se\mbed-cloud-client-example-internal-poc-atmel-se\mbed-os\tools\device_management.py init -d arm.com --model-name k64f"

       Tip: You could retry the last command with "-v" flag for verbose output
```

jenia81, 2020-01-16:
Also @teetak01 @yogpan01

teetak01, 2020-01-16:
@madchutney can you check this?

cmonr, 2020-01-16:
@GuyWi It would probably help to supply the versions of mbed-cli and mbed-os that you're using, along with the OS.

GuyWi, 2020-01-19:
mbed-cli version 1.10.0
mbed-os-5.15.0
Windows 10

madchutney, 2020-01-20:
It looks like the Pelion REST API is not returning any items in the collection which matches the API key configured to be used with Mbed. The error is far from informative but it is worth checking that the API key configured locally is available in Portal.