ARMmbed/mbed-cli #949: Auto-installing missing python modules fails to install cmsis-pack-manager

ygoyal18, 2020-01-22:
mbed-cli tries to install the missing packages 
```| [mbed] Updating library "mbed-cloud-client" to rev #4d7e04083520 (tag: 3.0.0)
| [mbed] Updating library "mbed-os" to rev #6bf7fd3ecc60 (tags: mbed-os-5.12.0, mbed-os-5.12.0-rc4)
| [mbed] Auto-installing missing Python modules (manifest_tool, cmsis_pack_manager)...
```

but then errors out. On manually installing the mbed-os requirements we get this - 

```
Installing collected packages: cmsis-pack-manager, ecdsa, pyasn1, pyparsing, asn1ate, protobuf, manifest-tool
    Running setup.py install for cmsis-pack-manager ... error
    ERROR: Command errored out with exit status 1:
     command: /home/ubuntu/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mbed-fcc/1.0+gitAUTOINC+8b3b41046c-r0/recipe-sysroot-native/usr/bin/python-native/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-9C5lRW/cmsis-pack-manager/setup.py'"'"'; __file__='"'"'/tmp/pip-install-9C5lRW/cmsis-pack-manager/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-bz7fK3/install-record.txt --single-version-externally-managed --compile --install-headers /home/ubuntu/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mbed-fcc/1.0+gitAUTOINC+8b3b41046c-r0/recipe-sysroot-native/usr/include/python2.7/cmsis-pack-manager
         cwd: /tmp/pip-install-9C5lRW/cmsis-pack-manager/
    Complete output (9 lines):
    running install
    running build
    running build_py
    creating build/lib
    creating build/lib/cmsis_pack_manager
    copying cmsis_pack_manager/pack_manager.py -> build/lib/cmsis_pack_manager
    copying cmsis_pack_manager/_version.py -> build/lib/cmsis_pack_manager
    copying cmsis_pack_manager/__init__.py -> build/lib/cmsis_pack_manager
    error: [Errno 2] No such file or directory
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/ubuntu/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mbed-fcc/1.0+gitAUTOINC+8b3b41046c-r0/recipe-sysroot-native/usr/bin/python-native/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-9C5lRW/cmsis-pack-manager/setup.py'"'"'; __file__='"'"'/tmp/pip-install-9C5lRW/cmsis-pack-manager/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-bz7fK3/install-record.txt --single-version-externally-managed --compile --install-headers /home/ubuntu/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/mbed-fcc/1.0+gitAUTOINC+8b3b41046c-r0/recipe-sysroot-native/usr/include/python2.7/cmsis-pack-manager Check the logs for full command output.```

The latest of cmsis-pack-manager is missing a _version.py file - https://github.com/ARMmbed/cmsis-pack-manager/tree/master/cmsis_pack_manager 

I feel that can't be the issue, as it was working for me last week. Not sure why it is looking for a _version.py file now.

cmonr, 2020-01-22:
@yogpan01 A bet. Try downgrading your `pip` module version.
I've seen some discussion recently w.r.t issues with its latest release.

acabarbaye, 2020-01-23:
@ygoyal18 : the problem is indeed due to `pip`. the latest release (20) forces a rebuild of the wheel which fails in different ways https://github.com/ARMmbed/cmsis-pack-manager/issues/137.
Please downgrade pip to 19 `pip install pip==19` and then the `mbed-cli` commands such as `mbed import` will start working fine again.

cmonr, 2020-01-23:
@acabarbaye Thanks for the confirmation.

This'll unfortunately be a bit tricky for us since the installation context for this is within a seperate tool (bitbake/yocto): https://github.com/armPelionEdge/meta-pelion-edge/blob/dev/recipes-wigwag/mbed-fcc/mbed-fcc_0.0.1.bb#L25

Since this isn't an mbed-cli issue, I'll be closing this ticket.

closed 2020-01-23 by cmonr