ARMmbed/mbed-cli #952: Mbed CLI `mbed` doesn't work

andrewc-arm, 2020-02-05:
Hi,

`mbed` command line used to work well but my `mbed` does not work any more. I tried reinstalling the setup installer https://github.com/ARMmbed/mbed-cli-windows-installer/releases/tag/v0.4.10 but it still does not work.
```
c:\mbed-cli\mbed-cli\mbed-cli-1.8.3>mbed
'mbed' is not recognized as an internal or external command,
operable program or batch file.

c:\mbed-cli\mbed-cli\mbed-cli-1.8.3>echo %PATH%
C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Mercurial;C:\Program Files\Git\cmd;C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.2\arm\bin;C:\Program Files\ARMCompiler6.12\bin;C:\Program Files\PuTTY\;C:\Program Files\nodejs\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\TortoiseGit\bin;C:\Program Files\Java\jdk-12.0.2\bin;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:\Program Files (x86)\GNU Tools ARM Embedded\6 2017-q2-update\bin;C:\Users\andcho01\AppData\Local\Programs\Python\Python37\Scripts\;C:\Users\andcho01\AppData\Local\Programs\Python\Python37\;C:\Program Files (x86)\GNU Tools ARM Embedded\7 2017-q4-major\bin;C:\Users\andcho01\AppData\Local\Microsoft\WindowsApps;C:\Users\andcho01\AppData\Roaming\npm

c:\mbed-cli\mbed-cli\mbed-cli-1.8.3>python
Python 3.7.6 (tags/v3.7.6:43364a7ae0, Dec 19 2019, 00:42:30) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>>
```
Could you please help me on this?
Thank you!

andrewc-arm, 2020-02-05:
Hi,
I was able to reproduce this problem from the new machine as well.
If I install the Mbed-CLI package and then upgrade to Python3, I get this failure.
So it seems I need to run this in Python 3 environment.
```
pip install mbed-cli
```
And then, I got this failure.
```
Collecting mercurial>=5.2 (from mbed-cli)
  Downloading https://files.pythonhosted.org/packages/91/3b/e531b6845dda605b3bb9835f2c260ed6917f2cbda1a999feb8da6632a697/mercurial-5.3.tar.gz (7.5MB)
     |████████████████████████████████| 7.5MB 1.6MB/s
    ERROR: Command errored out with exit status 1:
     command: 'c:\users\andcho01\appdata\local\programs\python\python38-32\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\andcho01\\AppData\\Local\\Temp\\pip-install-ubidtvz_\\mercurial\\setup.py'"'"'; __file__='"'"'C:\\Users\\andcho01\\AppData\\Local\\Temp\\pip-install-ubidtvz_\\mercurial\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base pip-egg-info
         cwd: C:\Users\andcho01\AppData\Local\Temp\pip-install-ubidtvz_\mercurial\
    Complete output (17 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\andcho01\AppData\Local\Temp\pip-install-ubidtvz_\mercurial\setup.py", line 1500, in <module>
        setup_zstd.get_c_extension(
      File "contrib/python-zstandard\setup_zstd.py", line 159, in get_c_extension
        compiler.initialize()
      File "c:\users\andcho01\appdata\local\programs\python\python38-32\lib\distutils\_msvccompiler.py", line 253, in initialize
        vc_env = _get_vc_env(plat_spec)
      File "c:\users\andcho01\appdata\local\programs\python\python38-32\lib\site-packages\setuptools\msvc.py", line 171, in msvc14_get_vc_env
        return EnvironmentInfo(plat_spec, vc_min_ver=14.0).return_env()
      File "c:\users\andcho01\appdata\local\programs\python\python38-32\lib\site-packages\setuptools\msvc.py", line 1075, in __init__
        self.si = SystemInfo(self.ri, vc_ver)
      File "c:\users\andcho01\appdata\local\programs\python\python38-32\lib\site-packages\setuptools\msvc.py", line 547, in __init__
        vc_ver or self._find_latest_available_vs_ver())
      File "c:\users\andcho01\appdata\local\programs\python\python38-32\lib\site-packages\setuptools\msvc.py", line 561, in _find_latest_available_vs_ver
        raise distutils.errors.DistutilsPlatformError(
    distutils.errors.DistutilsPlatformError: Microsoft Visual C++ 14.0 is required. Get it with "Build Tools for Visual Studio": https://visualstudio.microsoft.com/downloads/
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
WARNING: You are using pip version 19.2.3, however version 20.0.2 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.
```

Any chance we have Mbed CLI installer with Python 3 inside?

andrewc-arm, 2020-02-05:
OK. I found how I can resolve this.
This problem seems to be only existing in Windows only.
* I downloaded Visual Studio Professional evaluation version since my company is non-community.
* https://visualstudio.microsoft.com/ko/downloads/
* Install Visual C++ component for now.
* Then after Visual Studio installation, I ran `pip install mbed-cli` again and it installed well.
```
C:\Users\andcho01>mbed --version
1.10.2
```
Maybe, I can later uninstall Visual Studio after this installation.

closed 2020-02-05 by andrewc-arm

acabarbaye, 2020-03-09:
this is similar to https://github.com/ARMmbed/mbed-cli/issues/944#issuecomment-588963687