ARMmbed/mbed-cli #957: Mbed Cli not working with Python3

yasir-khilji-64, 2020-03-06:
I recently installed Ubuntu on my work PC and install Python3 Pip. Installed virtualenv and made a separate env for mbed and pip install mbed-cli. when I do mbed new `name-of-project`, it stucks at `[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at branch/tag "latest"`. I thought it was an issue with virtaulenv, and installed mbed-cli system wide, but still the same issue. It works on Python2. but we are changing our dev env from python2 to python3. Please help ASAP.

yasir-khilji-64, 2020-03-06:
[image]
Issue is here. 
please help.

acabarbaye, 2020-03-09:
Could you tell us the `git` version, you are using and also ensure that it is in the `PATH`?

acabarbaye, 2020-03-10:
I have tried to reproduce the problem but unfortunately I don't see any issue with python3:
```
adrcab01@E106283:~/testmbed$ python -V
Python 3.7.3
adrcab01@E106283:~/testmbed$ mbed new blink
[mbed] Working path "/home/adrcab01/testmbed" (directory)
[mbed] Creating new program "blink" (git)
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at branch/tag "latest"
[mbed] Updating reference "mbed-os" -> "https://github.com/ARMmbed/mbed-os/#e642a7d8b3609a7c903e042cd772f00a5d299088"
[mbed] Auto-installing missing Python modules (click)...
adrcab01@E106283:~/testmbed$ uname --all
Linux E106283 4.15.0-88-generic #88-Ubuntu SMP Tue Feb 11 20:11:34 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

closed 2020-03-14 by yasir-khilji-64