ARMmbed/mbed-cli #958: Link to https://github.com/ARMmbed/mbed-cli/blob/master/README.md is broken (404)

N0rbert, 2020-03-06:
You have broken link in your main [README.rst](https://github.com/ARMmbed/mbed-cli/blob/master/README.rst) file, i.e.

> `read the documentation on GitHub <https://github.com/ARMmbed/mbed-cli/blob/master/README.md>`

returns 404.
This link is also mentioned on [PyPi page of the library](https://pypi.org/project/mbed-cli/).

Please fix it and also provide actual link to the relevant documentation.

idea--list, 2020-03-21:
Yes, that is not a good boarding experience to the cli.
Under this link you can find the current doc:
https://github.com/ARMmbed/mbed-cli/blob/1.8.3/README.md

mark-edgeworth, 2020-04-14:
Fixed in https://github.com/ARMmbed/mbed-cli/pull/950

closed 2020-04-14 by mark-edgeworth