ARMmbed/mbed-cli #966: pypi version doesn't match mbed --version

thendrickson, 2020-03-30:
On Mac:

pypi.org shows version 1.10.3
"pip3 install mbed" says it successfully installs 1.10.3
/Library/Frameworks/Python.framework/Versions/3.8/bin/mbed --version reports 1.10.2

mstfkaratas, 2020-04-02:
I think they forgot to update version string on mbed.py
https://github.com/ARMmbed/mbed-cli/blob/c2528f6fca8d73be0f5cb5e494dd4987f40c39fa/mbed/mbed.py#L56

`$ pip3 list` 
command should show correct version.
```
....
mbed-cli           1.10.3     
mbed-cloud-sdk     2.0.8 
....
```

arvoelke, 2020-04-06:
Thanks for posting this issue. I spent a good chunk of time repeatedly uninstalling and reinstalling because I couldn't figure out why `mbed --version` said I was stuck on an older release. In case it helps, there appear to be two different version strings. One is the above, and the other is in `setup.py` (`__version__ = '1.10.3'`). The standard approach to preventing this is to have a single version string in `version.py` and then to import that string in all of the required places. e.g. code: https://github.com/scipy/scipy/blob/3fc8aec56d5378fd191dbcea294219147bd93ccc/setup.py#L101