ARMmbed/mbed-cli #968: Hanging on 'Downloading library build "*" (might take a while)'

arvoelke, 2020-04-06:
```
mbed new simple_test --mbedlib 
[mbed] Working path "/home/arvoelke/git/..." (program)
[mbed] Creating new program "simple_test" (git)
[mbed] Adding library "mbed" from "https://mbed.org/users/mbed_official/code/mbed/builds" at branch/tag "tip"
[mbed] Downloading library build "65be27845400" (might take a while)
```

I've tried this three times now, each time keeping it open for more than 24 hours -- and it still hangs.

If I visit the above url (https://mbed.org/users/mbed_official/code/mbed/builds) and click on the first build (`65be27845400`), it takes me to an invalid page:

https://os.mbed.com/users/mbed_official/code/mbed//builds/65be27845400/
> An error occurred while processing your request
> no such method: builds

Note there are two slashes (`//`) in the URL. If I remove one of the slashes then the link becomes valid. Is there a fix or work-around? Thanks.

TheMcSebi, 2022-12-03:
For everyone still encountering this issue after 2022:

    pip uninstall mbed-cli
    git clone https://github.com/ARMmbed/mbed-cli.git
    cd mbed-cli
    pip install --editable .

edit mbed/mbed.py and change line 397 to:

    while data != b'':

then you should be able to download the libraries

Maaajaaa, 2023-04-13:
`return stdout.decode("ANSI")`

led to this error on my machine (Ubuntu 22.04):

`ERROR: Unknown Error: unknown encoding: ANSI`

so I changed it back to the original, then it worked successfully.

DhariMT, 2024-02-18:
> For everyone still encountering this issue after 2022:
> 
> ```
> pip uninstall mbed-cli
> git clone https://github.com/ARMmbed/mbed-cli.git
> cd mbed-cli
> pip install --editable .
> ```
> 
> edit mbed/mbed.py and change line 397 to:
> 
> ```
> while data != b'':
> ```
> 
> and line 294 to:
> 
> ```
> return stdout.decode("ANSI")
> ```
> 
> then you should be able to download the libraries

This worked but without editing line 294 on Ubuntu (editing the line gave another error when creating the project). Thanks a lot, I have been going crazy over this