ARMmbed/mbed-cli #978: README doesn't link to documentation.

hpigot, 2020-07-03:
README.rst states "The best way to get started is to read the documentation on GitHub." but just links back to itself. https://github.com/ARMmbed/mbed-cli/blob/master/README.rst

Include link to documentation.

mnaberez, 2020-07-03:
> README.rst states "The best way to get started is to read the documentation on GitHub." but just links back to itself. https://github.com/ARMmbed/mbed-cli/blob/master/README.rst

The `README.rst` file is used as the [PyPI project description](https://pypi.org/project/mbed-cli/1.10.4/).  This is a common pattern with Python packages.  The link provides a way for PyPI users to get to the GitHub repository.

hpigot, 2020-07-13:
It would be helpful if the README also included a link to the latest documentation, which seems pretty non-existent in the master branch https://github.com/ARMmbed/mbed-cli/blob/master/docs/index.md Is there somewhere else users should look to find up to date documentation for mbed CLI?