ARMmbed/mbed-cli #986: Issue with version when creating a new program

Nakul93, 2020-11-03:
Whenever I try to create a new program using the mbed-cli command, mbed new 'new-proj'. The mbed-os version of the code that gets pulled is an old one i.e v5.15.4. Even when I try to update the master using mbed update master command it still doesn't update the mbed-os to latest version i.e 6.4.

Mbed-cli version = 1.10.4
Python version = 3.7.9
git version =2.29.2

JanneKiiskila, 2020-11-03:
What do the `.lib` files state? If the `.lib` files do not have an exact hash, well - the results are rather unpredictable.

Nakul93, 2020-11-03:
The mbed-os.lib file has this link- https://github.com/ARMmbed/mbed-os/#e4b81f67f939a0c0b11c147ce74aa367271e1279
Also noticed that the tag latest is on v5.15.4

JanneKiiskila, 2020-11-03:
Are you using mbed cache? That has had some issues.

Nakul93, 2020-11-03:
Yeah so that was the issue, I just disabled the cache using `mbed cache off` cmd before importing a new program and now it cloned the latest version 6.4.0. Before closing this issue, do I need to do this every time or this cache setting a universal thing.

JanneKiiskila, 2020-11-03:
The setting stays, if you turn it off (especially if you used the -G then it should be globally) if stays off.

Nakul93, 2020-11-03:
Thank you. Closing this now.

closed 2020-11-03 by Nakul93

JanneKiiskila, 2020-11-03:
It's probably still a bug, though. :-)

JanneKiiskila, 2020-11-03:
Also, on the tag - any notion of using a tag to identify `latest` is just blatant misuse of git tags - [detailed out here](https://git-scm.com/docs/git-tag#_on_re_tagging) in the official git documentation. There should have never been a tag called `latest`. There could have been a branch called latest, but not a tag. Tags are write once, they cannot and should not be updated/moving.