ARMmbed/mbed-cli #992: Error compiling simple blinky against NUCLEO_G031K8 target

Michael-Liao, 2021-05-26:
### System and Library versions
Ubuntu 20.04
Python 3.7.7
mbed-cli 1.10.5
mbed-os 6.4.0

### output
```bash
Building project mbed-os-example-blinky (NUCLEO_G031K8, GCC_ARM)
Scan: mbed-os-example-blinky
Macros: -DTARGET_STM32G031xx -DMBED_BUILD_TIMESTAMP=1622020309.7602131 -DDEVICE_SLEEP=1 -DDEVICE_PORTINOUT=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -DUSE_FULL_LL_DRIVER -DDEVICE_SPISLAVE=1 -DDEVICE_RTC=1 -DDEVICE_LPTICKER=1 -DDEVICE_MPU=1 -DDEVICE_SERIAL=1 -D__CMSIS_RTOS -DDEVICE_SPI=1 -DTARGET_CORTEX -DTARGET_MCU_STM32 -DTARGET_MCU_STM32G031xx -DDEVICE_RESET_REASON=1 -DTARGET_M0P -D__MBED__=1 -DDEVICE_WATCHDOG=1 -DTARGET_CORTEX_M -DSTM32G031xx -DDEVICE_USTICKER=1 -DDEVICE_SPI_ASYNCH=1 -DTARGET_NAME=NUCLEO_G031K8 -DMBED_TICKLESS -DDEVICE_FLASH=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SERIAL_FC=1 -DTARGET_LIKE_MBED -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DCOMPONENT_FLASHIAP=1 -DDEVICE_I2C=1 -D__CORTEX_M0PLUS -DUSE_HAL_DRIVER -DDEVICE_I2C_ASYNCH=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_MCU_STM32G0 -DTARGET_LIKE_CORTEX_M0 -DTARGET_NUCLEO_G031K8 -D__MBED_CMSIS_RTOS_CM -DTARGET_STM32G0 -DARM_MATH_CM0PLUS -DDEVICE_PORTIN=1 -DDEVICE_I2CSLAVE=1 -DDEVICE_PORTOUT=1 -DTARGET_STM -DTOOLCHAIN_GCC -DEXTRA_IDLE_STACK_REQUIRED -DTARGET_RELEASE -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_SERIAL_ASYNCH=1
Compile [ 54.1%]: ws_pae_time.c
Compile: /usr/bin/arm-none-eabi-gcc -c -std=gnu11 -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -ffunction-sections -fdata-sections -funsigned-char -MMD -fomit-frame-pointer -Os -g -DMBED_TRAP_ERRORS_ENABLED=1 -DMBED_RTOS_SINGLE_THREAD -D__NEWLIB_NANO -DMBED_MINIMAL_PRINTF -mcpu=cortex-m0plus -mthumb -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x10000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0x2000 -DTARGET_STM32G031xx -DMBED_BUILD_TIMESTAMP=1622020309.7602131 -DDEVICE_SLEEP=1 -DDEVICE_PORTINOUT=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -DUSE_FULL_LL_DRIVER -DDEVICE_SPISLAVE=1 -DDEVICE_RTC=1 -DDEVICE_LPTICKER=1 -DDEVICE_MPU=1 -DDEVICE_SERIAL=1 -D__CMSIS_RTOS -DDEVICE_SPI=1 -DTARGET_CORTEX -DTARGET_MCU_STM32 -DTARGET_MCU_STM32G031xx -DDEVICE_RESET_REASON=1 -DTARGET_M0P -D__MBED__=1 -DDEVICE_WATCHDOG=1 -DTARGET_CORTEX_M -DSTM32G031xx -DDEVICE_USTICKER=1 -DDEVICE_SPI_ASYNCH=1 -DTARGET_NAME=NUCLEO_G031K8 -DMBED_TICKLESS -DDEVICE_FLASH=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SERIAL_FC=1 -DTARGET_LIKE_MBED -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DCOMPONENT_FLASHIAP=1 -DDEVICE_I2C=1 -D__CORTEX_M0PLUS -DUSE_HAL_DRIVER -DDEVICE_I2C_ASYNCH=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_MCU_STM32G0 -DTARGET_LIKE_CORTEX_M0 -DTARGET_NUCLEO_G031K8 -D__MBED_CMSIS_RTOS_CM -DTARGET_STM32G0 -DARM_MATH_CM0PLUS -DDEVICE_PORTIN=1 -DDEVICE_I2CSLAVE=1 -DDEVICE_PORTOUT=1 -DTARGET_STM -DTOOLCHAIN_GCC -DEXTRA_IDLE_STACK_REQUIRED -DTARGET_RELEASE -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_SERIAL_ASYNCH=1 @./BUILD/NUCLEO_G031K8/GCC_ARM/.includes_1194b2bdfd9a4ca739e277494defb00a.txt -include ./BUILD/NUCLEO_G031K8/GCC_ARM/mbed_config.h -MD -MF BUILD/NUCLEO_G031K8/GCC_ARM/mbed-os/connectivity/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_pae_time.d -o BUILD/NUCLEO_G031K8/GCC_ARM/mbed-os/connectivity/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_pae_time.o ./mbed-os/connectivity/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_pae_time.c
[Error] ws_pae_time.c@174,103: expected ')' before 'PRIi64'
[Warning] ws_pae_time.c@174,22: spurious trailing '%' in format [-Wformat=]
[DEBUG] Return: 1
[DEBUG] Output: In file included from ./mbed-os/connectivity/libraries/nanostack-libservice/mbed-client-libservice/ns_trace.h:32,
[DEBUG] Output:                  from ./mbed-os/connectivity/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_pae_time.c:22:
[DEBUG] Output: ./mbed-os/connectivity/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_pae_time.c: In function 'ws_pae_current_time_set':
[DEBUG] Output: ./mbed-os/connectivity/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_pae_time.c:174:103: error: expected ')' before 'PRIi64'
[DEBUG] Output:   174 |             tr_error("FATAL: system time less than reference time or more than 12 months in future: %"PRIi64" reference time: %"PRIi64, system_time, current_time);
[DEBUG] Output:       |                                                                                                       ^~~~~~
[DEBUG] Output: ./mbed-os/platform/mbed-trace/include/mbed-trace/mbed_trace.h:143:79: note: in definition of macro 'tr_error'
[DEBUG] Output:   143 | #define tr_error(...)           mbed_tracef(TRACE_LEVEL_ERROR,   TRACE_GROUP, __VA_ARGS__)   //!< Print Error Message
[DEBUG] Output:       |                                                                               ^~~~~~~~~~~
[DEBUG] Output: ./mbed-os/connectivity/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_pae_time.c:174:22: warning: spurious trailing '%' in format [-Wformat=]
[DEBUG] Output:   174 |             tr_error("FATAL: system time less than reference time or more than 12 months in future: %"PRIi64" reference time: %"PRIi64, system_time, current_time);
[DEBUG] Output:       |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[DEBUG] Output: ./mbed-os/platform/mbed-trace/include/mbed-trace/mbed_trace.h:143:79: note: in definition of macro 'tr_error'
[DEBUG] Output:   143 | #define tr_error(...)           mbed_tracef(TRACE_LEVEL_ERROR,   TRACE_GROUP, __VA_ARGS__)   //!< Print Error Message
[DEBUG] Output:       |                                                                               ^~~~~~~~~~~
[DEBUG] Output: ./mbed-os/connectivity/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_pae_time.c:174:101: note: format string is defined here
[DEBUG] Output:   174 |             tr_error("FATAL: system time less than reference time or more than 12 months in future: %"PRIi64" reference time: %"PRIi64, system_time, current_time);
[DEBUG] Output:       |                                                                                                     ^
Traceback (most recent call last):
  File "/home/michael/Documents/PhD/arc/mbed-os-example-blinky/mbed-os/tools/toolchains/mbed_toolchain.py", line 559, in compile_queue
    res['command']
  File "/home/michael/Documents/PhD/arc/mbed-os-example-blinky/mbed-os/tools/toolchains/mbed_toolchain.py", line 684, in compile_output
    raise ToolException(stderr)
tools.utils.ToolException: In file included from ./mbed-os/connectivity/libraries/nanostack-libservice/mbed-client-libservice/ns_trace.h:32,
                 from ./mbed-os/connectivity/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_pae_time.c:22:
./mbed-os/connectivity/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_pae_time.c: In function 'ws_pae_current_time_set':
./mbed-os/connectivity/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_pae_time.c:174:103: error: expected ')' before 'PRIi64'
  174 |             tr_error("FATAL: system time less than reference time or more than 12 months in future: %"PRIi64" reference time: %"PRIi64, system_time, current_time);
      |                                                                                                       ^~~~~~
./mbed-os/platform/mbed-trace/include/mbed-trace/mbed_trace.h:143:79: note: in definition of macro 'tr_error'
  143 | #define tr_error(...)           mbed_tracef(TRACE_LEVEL_ERROR,   TRACE_GROUP, __VA_ARGS__)   //!< Print Error Message
      |                                                                               ^~~~~~~~~~~
./mbed-os/connectivity/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_pae_time.c:174:22: warning: spurious trailing '%' in format [-Wformat=]
  174 |             tr_error("FATAL: system time less than reference time or more than 12 months in future: %"PRIi64" reference time: %"PRIi64, system_time, current_time);
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./mbed-os/platform/mbed-trace/include/mbed-trace/mbed_trace.h:143:79: note: in definition of macro 'tr_error'
  143 | #define tr_error(...)           mbed_tracef(TRACE_LEVEL_ERROR,   TRACE_GROUP, __VA_ARGS__)   //!< Print Error Message
      |                                                                               ^~~~~~~~~~~
./mbed-os/connectivity/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_pae_time.c:174:101: note: format string is defined here
[... cut ...]
  174 |             tr_error("FATAL: system time less than reference time or more than 12 months in future: %"PRIi64" reference time: %"PRIi64, system_time, current_time);
      |                                                                                                     ^

[mbed] Working path "/home/michael/Documents/PhD/arc/mbed-os-example-blinky" (program)
```

I've tried to search for it but it appears that `PRIi64`  error is due to mbed not finding `<inttypes.h>`.
How do I solve this?

Michael-Liao, 2021-05-26:
after changing to the `mbed-os-example-blinky-baremetal`. it successfully compiled now and everything is working as expected.

closed 2021-05-26 by Michael-Liao

jeromecoutant, 2021-05-26:
Yes, only bare metal is supported for such small devices:
https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L2622-L2623